rhyst/linak-controller #3: No luck with Linak DPG1C

bradstcroix, 2020-08-12:
I've been trying a number of repo's on Github to get my Linak DPG1C working with no luck so far.

After getting this application up and running I'm able to successfully connect to the desk and can retrieve the desk height with no issue.

However when using the -sit command nothing happens and the python script sits there waiting.

I've had the same issue with other Idasen applications from Github. I'm guessing maybe my Linak controller must be using a different payload or something, even though the service and characteristic UUID's match exactly to what's in the code.

Is there a way to see some verbose logging of what's happening when Gatt sends the up/down commands to the BT controller?

Even better, can you explain how I might be able to sniff the packets that control it using the official Linak app (as that works without issue)?

Thanks for any assistance

bradstcroix, 2020-08-12:
Update: After some messing around with wireshark and the Linak Desk Connect app on Android I found that the commands are exactly the same on the DPG1C controller. So it wasn't that.

After factory resetting the desk and re-pairing using bluetoothctl, I'm pleased to say it now works!

Thanks for the great app!

rhyst, 2020-08-12:
Nice one, glad you got it working :+1:

closed 2020-08-12 by rhyst