rhyst/linak-controller #4: No Windows compatibility?

aienabled, 2020-10-31:
Hello,
as far as I've researched python3-dbus is not available for Windows as it's intended only for Linux and Mac. I wish to know if there is a workaround. If not, perhaps it would be best to mention this in the README file.

Regards!

rhyst, 2020-10-31:
Hey,

Thanks for the suggestion, I'll make it clear that it currently only works for Linux/Mac in the README.

I might also have a play with getting it running on Windows, so I will leave this issue open for now. I will need to swap in a different Bluetooth library, maybe I can get on running that works on all platforms.

rhyst, 2020-11-12:
I've made a branch `bleak` (In progress PR #6) that switches the BLE library from `gatt-python` to `bleak` which should work on Linux, Windows, Mac.

My experience with it so far is it seems to work quite well on Windows, but seems flakey on Linux. It seems to often leave the connection in a bad state so it can't reconnect. I'm not used to `asyncio` though so it's possible I'm not using it correctly.

I will continue playing around with it but you can try the `bleak` branch on Windows now, feedback welcome.

aienabled, 2020-11-13:
Thank you very much, @rhyst!
I've tried `bleak` branch and can confirm that it works fine under Windows.
Some notes:
- I've used [Bluetooth LE Explorer](https://www.microsoft.com/en-us/p/bluetooth-le-explorer/9n0ztkf1qd98) to locate and pair with the desk (Windows pairing started when I selected the Desk in the devices list) and to obtain the desk MAC address. I had to disconnected from the smartphone app first otherwise the desk was not discoverable by the app (as expected by Bluetooth LE).
- I've used an external USB Adapter: TP-Link UB400 Bluetooth 4.0 Nano USB Adapter.
- I've used Python 3.8.6 as Twisted doesn't support 3.9.0 yet. Also, for some reason I have to use `python` instead of `python3`.

So now it works fine and I can drive the desk to the desired height.
I've created a separate ticket regarding the drive precision problem https://github.com/rhyst/idasen-controller/issues/7

Regards!

rpdswtk, 2020-11-13:
Hi @rhyst 
I created a scan feature which works with `--scan ADAPTER_NAME` parameter. I used qatt module for this.
If you like this feature let me know and I'll rewrite it with bleak

rhyst, 2020-11-13:
@aienabled Thanks! I'll add a set up / troubleshooting page I think to record useful tips and add. Possibly a "recipes" dir as well for things like your autohotkey setup and my albert config.

@rpdswtk Very happy to accept a PR. And if you have feedback on if the bleak branch works on Linux for you that would be very helpful and useful to decide whether I want to switch everything to bleak or maintain two different scripts. Bleak seems a bit slower to connect and I'm not 100% confident in the disconnect :grimacing:

rpdswtk, 2020-11-14:
Started testing using Ubuntu 20.04 with no connection issue, but feels definitely slower.

zeroarst, 2020-11-19:
I am a Windows user and not really familiar with python...
I try to use it with bleak branch. 
I have to manually install bruez because when I run python3 main.py it gives me error cannot find bluetoothctl.
Also there is a typo on README.md. blueoothctl > bluetoothctl

rhyst, 2020-11-20:
Hey @zeroarst, have you tried using the repo in Windows without WSL? The fact that you are having issues in `bluetoothctl` suggests this is beyond the scope of this script to fix, I have no idea if Bluez works at all in WSL.

rhyst, 2020-11-20:
I've pushed a small change that works around a bug where bleak does not use the specified adapter when scanning/connecting. This allows me to use a different Bluetooth adapter which seems to work more consistently. 

It also allows you to specify a `scan_timeout` which I have found works when set to `1` and makes it feel much quicker. If it keeps working for me I think I'll merge it soon.

zeroarst, 2020-11-26:
> 
> 
> Thank you very much, @rhyst!
> I've tried `bleak` branch and can confirm that it works fine under Windows 10 (1909).
> Some notes:
> 
>     * I've used [Bluetooth LE Explorer](https://www.microsoft.com/en-us/p/bluetooth-le-explorer/9n0ztkf1qd98) app from Microsoft to locate and pair with the desk (Windows pairing started when I selected the Desk in the devices list) and to obtain the desk MAC address. I had to disconnect from the smartphone app first otherwise the desk was not discoverable by the app (as expected by Bluetooth LE).
> 
>     * I've used an external USB Adapter: TP-Link UB400 Bluetooth 4.0 Nano USB Adapter.
> 
>     * I've used Python 3.8.6 as Twisted doesn't support 3.9.0 yet. Also, for some reason I have to use `python` instead of `python3`.
> 
> 
> So now it works fine and I can drive the desk to the desired height.
> I've created a separate ticket regarding the drive precision problem #7 .
> 
> Regards!
> 
> UPD. If anyone curious, here is my AutoHotkey script to drive the desk to stand and sit mode by pressing Ctrl+Alt+Shift+Up or Down arrow respectively: (you can change arrows to PgUp PgDn but in my case, these shortcuts are already taken for other actions)
> 
> ```
> ;Idasen Desk - Stand
> ^!+Up::Run "C:\Users\...\Desk - Stand.lnk"
> 
> ;Idasen Desk - Sit
> ^!+Down::Run "C:\Users\...\Desk - Sit.lnk"
> ```
> 
> (I've created shortcut files on the desktop but it's not necessary and could be simple `python` calls)

May I ask you what did you get it working on Windows, through the Windows Command line?

rhyst, 2020-11-27:
Merged the bleak branch with master.

closed 2020-11-27 by rhyst