rhyst/linak-controller #9: Error in bleak: invalid syntax

rhyst, 2020-11-27:
@aienabled 

May I ask you what did you get it working on Windows, through the Windows Command line?

I tried to use git bash seem better but still getting errors.... I am not familiar with python it seems the exception is from colon symbol...variable cannot be declared with type..?
my python version
[image]

When I run python main.py
[image]

And I looked at the source code of bleak...
[image]

Really want to get this working on Windows.

_Originally posted by @zeroarst in https://github.com/rhyst/idasen-controller/issues/4#issuecomment-734132244_

rhyst, 2020-11-27:
@zeroarst are you able to update Python? I'm on 3.9. That's the only thing I could think of. 3.5.1 should be able to handle type hinting but maybe there's some newer syntax in there as well.

zeroarst, 2020-11-29:
I tried to upgrade 3.9. dose not seem to see that error again.. but I am getting another error when run `pip install -r requirements.txt`
```

 ERROR: Command errored out with exit status 1:
     command: 'c:\users\roy\appdata\local\programs\python\python39\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Roy\\AppData\\Local\\Temp\\pip-install-qb52u9m1\\pythonnet\\setup.py'"'"'; __file__='"'"'C:\\Users\\Roy\\AppData\\Local\\Temp\\pip-install-qb52u9m1\\pythonnet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Roy\AppData\Local\Temp\pip-record-t9iawqu4\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\roy\appdata\local\programs\python\python39\Include\pythonnet'
         cwd: C:\Users\Roy\AppData\Local\Temp\pip-install-qb52u9m1\pythonnet\
    Complete output (6 lines):
    usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: setup.py --help [cmd1 cmd2 ...]
       or: setup.py --help-commands
       or: setup.py cmd --help

    error: option --single-version-externally-managed not recognized
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\roy\appdata\local\programs\python\python39\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Roy\\AppData\\Local\\Temp\\pip-install-qb52u9m1\\pythonnet\\setup.py'"'"'; __file__='"'"'C:\\Users\\Roy\\AppData\\Local\\Temp\\pip-install-qb52u9m1\\pythonnet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Roy\AppData\Local\Temp\pip-record-t9iawqu4\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\roy\appdata\local\programs\python\python39\Include\pythonnet' Check the logs for full command output.
```

Googled the issue: `--single-version-externally-managed not recognized`, happens when installing Twisted, found manually install wheel might solve the problem, which I did. However, getting another error.
`error: Microsoft Visual C++ 14.0 is required. Get it with "Build Tools for Visual Studio": https://visualstudio.microsoft.com/downloads/`

I think I will try to solve it another day..

MeTimesThree, 2020-12-07:
As aienabled poimted out [here](https://github.com/rhyst/idasen-controller/issues/4#issuecomment-726769913) python version 3.8.6 seems to be your best bet. I used the master branch and got it working.

zeroarst, 2020-12-11:
I had some progress today...
- Installed python 3.8.6
- Install Microsoft Visual Studio because when ran pip install -r requirement.txt I got error: Microsoft Visual C++ 14.0 is required. Get it with "Build Tools for Visual Studio": https://visualstudio.microsoft.com/downloads/

Now I can finally run `python main.py`
but it says 
**Scanning - Desk ec:8e:1a:1b:83:4c Not Found
Could not find desk ec:8e:1a:1b:83:4c**

I found my desk address by [suggested Bluetooth LE Explorer here](https://github.com/rhyst/idasen-controller/issues/4#issuecomment-726769913), which is ec:8e:1a:1b:83:4c
[image]

run `python main.py --scan` I get:
$ python main.py --scan
Found 8 devices using default adapter
59:1A:9B:44:34:04: Unknown
EC:8E:1A:1B:83:4C: Unknown
74:BA:1B:C4:DD:DB: Apple, Inc. (b'\x10\x06\n\x1dYUH\x08')
43:7D:21:6F:F6:3C: Apple, Inc. (b'\x0c\x0e\x00\xe3|\xa0\xd7\x19\xdb\xc1\x06\xe9\x9c8\xbe\xa6')
F0:18:98:23:3D:33: Apple, Inc. (b'\x10\x05I\x1c\xd3\xe5a')
60:95:C6:1F:E9:EA: Apple, Inc. (b'\x0c\x0e\x00\xe3|\xa0\xd7\x19\xdb\xc1\x06\xe9\x9c8\xbe\xa6')
40:CD:B2:4B:4B:0E: Apple, Inc. (b'\x07\x19\x01\x0b +\x99\x8f\x00\x11\x05A\xae\xa1\xe8J/l\xef\xd5\x9eJ]\x83d\xac\xd1')
78:BB:F7:64:BD:26: Apple, Inc. (b'\x10\x07`\x1f\xa5\x01\xd6*(')

My config.yaml
mac_address: ec:8e:1a:1b:83:4c
stand_height: 1100
sit_height: 720
height_tolerance: 2.0
scan_timeout: 5
connection_timeout: 10
adapter_name: hci0

MeTimesThree, 2020-12-12:
I think the mac_address needs to be uppercase.

MeTimesThree, 2020-12-12:
On my first try it refused to connect and i had to hold the idasens bluetooth button for the entire duration of the search (thus wipeing the desk height-presets).

rhyst, 2021-01-12:
@MeTimesThree  What are the desk height presets you're referring to?

closed 2021-01-26 by rhyst

MeTimesThree, 2021-02-01:
By desk height-presets i mean those saved positions from the app that the desk stops at when using the hw-controller. Pressing the button for about eight seconds wipes them, but i think pressing the button for four seconds is enough and keeps them.

rhyst, 2021-02-01:
I think you might have a slightly different desk model? Is it this one: https://www.ikea.com/gb/en/p/idasen-desk-sit-stand-black-dark-grey-s89280993/ ?