rhyst/linak-controller #10: Bleak - Desk not Found

jeanblanchard, 2020-12-02:
After upgrading to the latest `master` with Bleak, my desk is not found any more. Not sure what I can do to help investigating.

## Steps
```bash
$ git checkout master
$ python3 main.py
Scanning - Desk FB:B4:B9:72:4C:F9 Not Found
Could not find desk FB:B4:B9:72:4C:F9

$ git checkout nobleak # commit defc47 with overridden mac_address in config.yaml
$ python3 main.py
Current height: 724.2mm
```

## Environment
* Linux (Solus 4.1)
* Python 3.7.7
* bleak-0.9.1

rhyst, 2020-12-07:
I has similar problems with the in build adaptor in my laptop. I started using an external adaptor instead. I do not know why `gatt-python` works and `bleak` does not as I think they both use the same sort of backend. It might be that I'm using `bleak` wrong but I did try for quite a while to get my in built adaptor to work so :man_shrugging: 

There's not really been much added to the `bleak` version so I suppose you could continue to use the old one. I might add a note about this to the README. Sorry I can't be more help.

rhyst, 2020-12-07:
I just had this problem and noticed that my adaptor names had switched round (`hci0` became `hci1` and vice versa) so possibly just check that the desk is definitely paired with the adaptor you've specified in the config?

jeanblanchard, 2020-12-08:
Well... It seems to be working now...

After playing around with bleak for a while and not getting anywhere, today I tried the [scanner](https://github.com/hbldh/bleak/blob/develop/examples/scanner.py) example script, and suddenly the "connect" works. Not sure if the scan filled a cache of some sort, or if it's totally unrelated (a reboot or package update maybe ?). But anyway all seems good now. Thanks for taking the time to look into it.

closed 2020-12-08 by jeanblanchard