rhyst/linak-controller #12: Keep connection alive

MeTimesThree, 2021-01-09:
Hi,

is there a way to always keep the desk connected so that move-commands can be sped up?

rhyst, 2021-01-12:
This person forked the repo to add a http interface: https://github.com/narthollis/idasen-controller

I think it'd need that or something similar so that you could set up the connection and keep it running but still accept commands.

That is a good idea for working around the lack of pickling. I might look into this but no promises :D

noxhirsch, 2021-01-23:
Oh yes, keeping the connection alive would be a really great addition. At the moment it takes about 10 seconds on my Win10 machine before the desk starts moving (but it *is* moving - so thank you very much for this!).
Maybe a socket connection is a good alternative to a http interface (https://stackoverflow.com/a/23149098).
Then main.py could connect to the table in the background and for example command.py would send the commands.

rhyst, 2021-01-25:
I've had a go at making a server to keep the connection alive on a branch [server-test](https://github.com/rhyst/idasen-controller/tree/server-test).

To use it start a server with:
```python3 main.py --server```
And then send commands with 
```python3 main.py --forward --<command>```
 where `<command>` is one of `sit`, `stand`, `move-to`.

Seems to work. On Windows it raises some random exceptions which I just catch and `pass` but it doesn't seem to care so 🤷 

Let me know if this improves things for you.

noxhirsch, 2021-01-26:
Wow, that was quick - thank you very much!

For me it works perfectly fine (just a few random exeptions as you said)
The server also seems to handle unpluggin the Bluetooth usb stick and standby very well :)

closed 2021-01-31 by rhyst

MeTimesThree, 2021-02-01:
Here i am bothering with other repos instead of keeping track of this post.
The new branch works brilliantly, thanks!