rhyst/linak-controller #13: Can't adjust height with physical button after interrupting script

backslashV, 2021-01-14:
Hi,
If in the middle of the script doing sit or stand I press the physical button on the desk, the desk stops moving and any subsequent button press only moves the desk a few milliliters and then stops.

rhyst, 2021-01-24:
Thanks for the report! I could not exactly reproduce this (I was able to move the desk with the button no matter what) but it was true that the script would hang if movement was interrupted by a button press. 

I've updated the script with a timeout and also an attempt to detect when a button press happens (it reported speed = 0 for me) and stop the script in both case. Let me know if that fixes the issue for you.

noxhirsch, 2021-01-26:
I could reproduce this issue and for me it's fixed now. Thank you very much! :)

rhyst, 2021-01-26:
Awesome, I'll close it then :+1:

closed 2021-01-26 by rhyst

backslashV, 2021-01-30:
Appreciate it.