rhyst/linak-controller #14: Installation not working on Raspberry Pi (can't find requirements.txt)

tomsturges, 2021-02-09:
Hi, thanks a lot for putting this script together, looking forward to using it.

On my Raspberry Pi, I first tried to install by typing `pip install idasen-controller`, however I get the error shown below

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting idasen-controller
  Downloading https://files.pythonhosted.org/packages/cd/d8/61d788f1661b5f2664ca0026d30d76d6745ffeb7ff55a4c30a95d5fd8f5d/idasen-controller-1.0.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-hIWMOj/idasen-controller/setup.py", line 4, in <module>
        with open('requirements.txt') as f:
    IOError: [Errno 2] No such file or directory: 'requirements.txt'
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-hIWMOj/idasen-controller/
```

How should I proceed?

Many thanks,
Tom

rhyst, 2021-02-09:
Can you try using python3?
```
pip3 install idasen-controller
```
I have not tested anything with python2 but I just ran this with `pip3` on my pi and it installed.

tomsturges, 2021-02-09:
Thanks a lot. It's proceeded further now. It looks like it installs everything correctly; however when I type `idasen-controller` I get the error `-bash: idasen-controller: command not found`

Here's the full installation log file if it's useful:

```
pi@raspberrypi:~ $ pip3 install idasen-controller
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting idasen-controller
  Downloading https://files.pythonhosted.org/packages/92/a5/341867037c4201b59089b0b2c56cc3984d1f973f513b74f7b6c986680926/idasen_controller-1.0.2-py3-none-any.whl
Collecting appdirs==1.4.4 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/3b/00/2344469e2084fb287c2e0b57b72910309874c3245463acd6cf5e3db69324/appdirs-1.4.4-py2.py3-none-any.whl
Collecting PyYAML==5.4.1 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/a0/a4/d63f2d7597e1a4b55aa3b4d6c5b029991d3b824b5bd331af8d4ab1ed687d/PyYAML-5.4.1.tar.gz (175kB)
    100% |████████████████████████████████| 184kB 1.5MB/s
  Installing build dependencies ... done
Collecting attrs==20.3.0 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/c3/aa/cb45262569fcc047bf070b5de61813724d6726db83259222cd7b4c79821a/attrs-20.3.0-py2.py3-none-any.whl (49kB)
    100% |████████████████████████████████| 51kB 2.7MB/s
Collecting zope.interface==5.2.0 (from idasen-controller)
  Downloading https://www.piwheels.org/simple/zope-interface/zope.interface-5.2.0-cp37-cp37m-linux_armv7l.whl (225kB)
    100% |████████████████████████████████| 235kB 1.4MB/s
Collecting incremental==17.5.0 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/f5/1d/c98a587dc06e107115cf4a58b49de20b19222c83d75335a192052af4c4b7/incremental-17.5.0-py2.py3-none-any.whl
Collecting six==1.15.0 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/ee/ff/48bde5c0f013094d729fe4b0316ba2a24774b3ff1c52d924a8a4cb04078a/six-1.15.0-py2.py3-none-any.whl
Collecting hyperlink==21.0.0 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/6e/aa/8caf6a0a3e62863cbb9dab27135660acba46903b703e224f14f447e57934/hyperlink-21.0.0-py2.py3-none-any.whl (74kB)
    100% |████████████████████████████████| 81kB 3.6MB/s
Collecting bleak==0.10.0 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/16/37/233041fca944da9eec5e80860c016be5ce42d6532811cbd81bdc889df682/bleak-0.10.0-py2.py3-none-any.whl (103kB)
    100% |████████████████████████████████| 112kB 2.6MB/s
Collecting Automat==20.2.0 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/dd/83/5f6f3c1a562674d65efc320257bdc0873ec53147835aeef7762fe7585273/Automat-20.2.0-py2.py3-none-any.whl
Collecting txdbus==1.1.2 (from idasen-controller)
  Downloading https://www.piwheels.org/simple/txdbus/txdbus-1.1.2-py2.py3-none-any.whl (45kB)
    100% |████████████████████████████████| 51kB 661kB/s
Collecting constantly==15.1.0 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/b9/65/48c1909d0c0aeae6c10213340ce682db01b48ea900a7d9fce7a7910ff318/constantly-15.1.0-py2.py3-none-any.whl
Collecting PyHamcrest==2.0.2 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/40/16/e54cc65891f01cb62893540f44ffd3e8dab0a22443e1b438f1a9f5574bee/PyHamcrest-2.0.2-py3-none-any.whl (52kB)
    100% |████████████████████████████████| 61kB 4.0MB/s
Collecting idna==3.1 (from idasen-controller)
  Downloading https://files.pythonhosted.org/packages/29/88/c52aae187d3b128a0f13f36a6c987fc0d408d03a678ad9996516925d8495/idna-3.1-py3-none-any.whl (58kB)
    100% |████████████████████████████████| 61kB 3.3MB/s
Collecting Twisted==20.3.0 (from idasen-controller)
  Downloading https://www.piwheels.org/simple/twisted/Twisted-20.3.0-cp37-cp37m-linux_armv7l.whl (3.1MB)
    100% |████████████████████████████████| 3.1MB 148kB/s
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from zope.interface==5.2.0->idasen-controller) (40.8.0)
Building wheels for collected packages: PyYAML
  Running setup.py bdist_wheel for PyYAML ... done
  Stored in directory: /home/pi/.cache/pip/wheels/2a/d4/92/cf299bdf4162957ca8126b46e913e29f76a4f17ca762c45028
Successfully built PyYAML
Installing collected packages: appdirs, PyYAML, attrs, zope.interface, incremental, six, idna, hyperlink, Automat, PyHamcrest, constantly, Twisted, txdbus, bleak, idasen-controller
  The script automat-visualize is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The scripts cftp, ckeygen, conch, mailmail, pyhtmlizer, tkconch, trial, twist and twistd are installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The script bleak-lescan is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The script idasen-controller is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed Automat-20.2.0 PyHamcrest-2.0.2 PyYAML-5.4.1 Twisted-20.3.0 appdirs-1.4.4 attrs-20.3.0 bleak-0.10.0 constantly-15.1.0 hyperlink-21.0.0 idasen-controller-1.0.2 idna-3.1 incremental-17.5.0 six-1.15.0 txdbus-1.1.2 zope.in
```

rhyst, 2021-02-09:
So it looks like pip is telling you what's wrong there:
```
Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
The script idasen-controller is installed in '/home/pi/.local/bin' which is not on PATH.
```
Either add `/home/pi/.local/bin` to your path or you can run it directly `/home/pi/.local/bin/idasen-controller`.

tomsturges, 2021-02-09:
Yup, sorry just figured that out a minute ago. Noob here!

It seems to be working now. Although there is one warning which I think I can fix on my own. I'll paste below just for reference

```
pi@raspberrypi:~ $ /home/pi/.local/bin/idasen-controller
:0: UserWarning: You do not have a working installation of the service_identity module: 'No module named 'service_identity''.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Scanning - Desk AA:AA:AA:AA:AA:AA Not Found
Could not find desk AA:AA:AA:AA:AA:AA
```

rhyst, 2021-02-09:
No problem at all 😊

It looks like I might need to add `service_identity` to the requirements, thanks for the heads up. I'll also update the readme to make `python3` a requirement, but I'll close this issue for now.

closed 2021-02-09 by rhyst