rhyst/linak-controller #15: Desk not found by idasen-controller (but paired with bluetooth)

tomsturges, 2021-02-10:
Hi again!

So my Raspberry Pi is paired to my desk but when I type `idasen-controller` I get the error  `Could not find desk e0:83:79:fd:a4:52` (I added it to the config.yaml). I confirmed that I am paired by typing `bluetoothctl` at the command line to open the Bluetooth Controller and then typed `info e0:83:79:fd:a4:52` which tells me:

```
Device E0:83:79:FD:A4:52 (random)
        Name: Standing Desk
        Alias: Standing Desk
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (99fa0001-338a-1024-8a49-009c0215f78a)
        UUID: Vendor specific           (99fa0010-338a-1024-8a49-009c0215f78a)
        UUID: Vendor specific           (99fa0020-338a-1024-8a49-009c0215f78a)
        UUID: Vendor specific           (99fa0030-338a-1024-8a49-009c0215f78a)
```

rhyst, 2021-02-10:
Raspberry Pis are weird, mine won't let me do bluetooth things as a non-root user so maybe that has something to do with it. If you install idasen controller as root:
```
sudo pip3 install idasen-controller
```
And then run the script as root:
```
sudo idasen-controller
```
Does that change anything?

tomsturges, 2021-02-11:
Thanks, but no. I did as you suggested (and re-connected in `bluetoothctl` as it had disconnected overnight) but it still cannot find the desk.

rhyst, 2021-02-11:
Ah so it seems that the mac_address must be uppercase. I have made the script uppercase it when reading from the config. Please update and let me know if that starts working (I think there is no need for sudo as your bluetoothctl works without it):

```
 pip3 install --upgrade idasen-controller
```

tomsturges, 2021-02-11:
I notice that update capitalising the MAC address now, but unfortunately I still have the same error (I checked with sudo too).

rhyst, 2021-02-11:
Is the mac address capitalised in the error?

tomsturges, 2021-02-11:
Yes it is.

rhyst, 2021-02-11:
Does running `idasen-controller --scan` return anything?

rhyst, 2021-02-11:
Also what python version are you running on the pi?

tomsturges, 2021-02-11:
```
pi@raspberrypi:~ $ python --version
Python 2.7.16
pi@raspberrypi:~ $ python3 --version
Python 3.7.3
```

tomsturges, 2021-02-11:
`idasen-controller --scan` returns a list of 10 devices, but none of them are the MAC address that `bluetoothctl` tells me I am connected to (and has the Alias "Standing Desk")

rhyst, 2021-02-11:
Very confusing! Whats the output of `hcitool dev`? Wondering if you somehow have two bluetooth adapters or something.

tomsturges, 2021-02-11:
```
pi@raspberrypi:~ $ hcitool dev
Devices:
        hci0    DC:A6:32:D3:FE:2D
```

tomsturges, 2021-02-11:
Btw I could not connect to my desk with the Linak android app, so I disconnected my RPi using `remove` inside `bluetoothctl` and now I can connect again. So I'm pretty sure that's the correct MAC address.

tomsturges, 2021-02-11:
Ahah! Now that I disconnected via `bluetoothctl`, the desk is now found with `idasen-controller --scan`. However I'm still having trouble connecting. I'll play a little bit and report back.

rhyst, 2021-02-11:
I've got everything basically the same as you:

```
➜  ~ python3 --version
Python 3.7.3
```
```
[bluetooth]# info E8:5B:5B:24:22:E4
Device E8:5B:5B:24:22:E4 (random)
	Name: Desk 8932
	Alias: Desk 8932
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (99fa0001-338a-1024-8a49-009c0215f78a)
	UUID: Vendor specific           (99fa0010-338a-1024-8a49-009c0215f78a)
	UUID: Vendor specific           (99fa0020-338a-1024-8a49-009c0215f78a)
	UUID: Vendor specific           (99fa0030-338a-1024-8a49-009c0215f78a)
```

```
➜  ~ hcitool dev
Devices:
	hci0	B8:27:EB:77:8C:03
```

```
➜  ~ idasen-controller --scan
Found 5 devices using hci0
E8:5B:5B:24:22:E4: Desk 8932
...
```
Except mine connects:
```
➜  ~ idasen-controller       
Connected E8:5B:5B:24:22:E4
Height:  688mm
Disconnected   
```
You could try adjusting the scan/connect timeouts and see if that helps? (I have noticed another bug that setting scan timeout from command line will not work so do that one in the config file).

tomsturges, 2021-02-11:
Ok thanks I'll report back in a couple hours, need to head out.

tomsturges, 2021-02-11:
Ok it works now! The (or one of the) problems was this...

I paired in `bluetoothctl` using `pair <MAC ADDRESS>`. It seems this would pair me, but also connect me, to the desk. So then `idasen-controller` was unable to connect. So to fix I used `disconnect <MAC ADDRESS>` in `bluetoothctl` so that I am paired but not connected. 

Perhaps you could update the README as it says "the desk should be connected and paired to the computer", whereas (at least with the RPi) it looks like it needs to be paired but NOT connected. Just a suggestion.

Thanks a lot for you help!

rhyst, 2021-02-11:
Glad its working. Interestingly on my pi it does not seem to matter if its connected in bluetoothctl or not but I will add it as a troubleshooting suggestion as you suggest.

closed 2021-02-11 by rhyst

magicmatt007, 2021-09-27:
Thanks for this Python code. It's the first one, which allowed me to control my desk.

I'd like to integrate this now into Homeassistant. However, before I start with this adventure, I'd need to resolve the following:

I only was able to use your code after I paired the desk in my Raspian on a PI3B. WIhtout pairing, I don't get it working at all.
However, I don't now, how I could automate pairing in Python......

Any idea, how I could get your code work without prior pairing? Or alternatively, do you have a proposal, how pairing could be done with Python code?

Many thanks!

rhyst, 2021-09-27:
A couple of ways you could:
- This uses the [bleak](https://bleak.readthedocs.io/en/latest/index.html) library to handle the Bluetooth stuff and I think that can pair with devices. You could either do that in a separate script or I'd definitely accept a PR to add pairing. There don't seem to be any examples of pairing code [but the interfaces do have a pair method ](https://bleak.readthedocs.io/en/latest/api.html#bleak.backends.client.BaseBleakClient.pair)so I expect it does something😬 
- If you know its running on a raspberry pi (and therefore is linux) then you can call the `bluetoothctl` command from python using the `os` or `subprocess` modules.