rhyst/linak-controller #17: Updated config.yaml but values aren't updated

backslashV, 2021-03-31:
Hi,
I have updated the config.yaml sit/stand height values but the script still uses the old values. I don't think these values are cached. Can you help please?

Thanks

backslashV, 2021-04-01:
I had to remove the config.yaml located at ~/.config/idasen-controller and then the values from the example/config.yaml folder were used. Is this expected? meaning the config.yaml at ~/.config/idasen-controller isn't automatically updated when example/config.yaml gets updated?

rhyst, 2021-04-01:
It is not expected that the `example/config.yaml` file is used. How have you installed it? 

I would recommend installing it using pip:

```
pip3 install idasen-controller
```

rhyst, 2021-04-01:
I just reread your comment. You should edit the values in the ` ~/.config/idasen-controller/config.yaml` file and the script should use that. 

You should not edit `example/config.yaml`. The only time that is used is if there is no config file in ~/.config/idasen-controller/` in which case it is copied over so you have something to edit.

backslashV, 2021-04-02:
Cool, thanks.

closed 2021-04-02 by backslashV