rhyst/linak-controller #19: Characteristic 99fa0002-338a-1024-8a49-009c0215f78a was not found

bluecmd, 2021-04-14:
Hello, thanks for your work!

I am trying to use your program on my IKEA IDÅSEN desk, but it seems to not like something with it.
Having looked at the bluetooth services provided, `99fa0002-338a-1024-8a49-009c0215f78a` is not one of them.

Am I running some odd firmware or something?

```
$ idasen-controller --mac-address 58:cb:52:4d:4a:4c
Connected 58:CB:52:4D:4A:4C
Traceback (most recent call last):
  File "/home/bluecmd/.local/bin/idasen-controller", line 8, in <module>
    sys.exit(init())
  File "/home/bluecmd/.local/lib/python3.8/site-packages/idasen_controller/main.py", line 416, in init
    asyncio.run(main())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/bluecmd/.local/lib/python3.8/site-packages/idasen_controller/main.py", line 410, in main
    await stop(client)
  File "/home/bluecmd/.local/lib/python3.8/site-packages/idasen_controller/main.py", line 177, in stop
    await client.write_gatt_char(UUID_COMMAND, COMMAND_STOP)
  File "/home/bluecmd/.local/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 587, in write_gatt_char
    raise BleakError("Characteristic {0} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 99fa0002-338a-1024-8a49-009c0215f78a was not found!
```

Thanks!

bluecmd, 2021-04-14:
Ah, I connected to the wrong device - my bad! :D :facepalm:

closed 2021-04-14 by bluecmd

rhyst, 2021-04-14:
😂 glad you sorted it!