rhyst/linak-controller #21: windows installation

eugenekostylev, 2021-05-23:
Hey there, any ideas how to fix this?

C:\Users\Eugene-PC>pip3 install idasen-controller
Collecting idasen-controller
  Using cached idasen_controller-1.0.5-py3-none-any.whl (11 kB)
Requirement already satisfied: attrs==20.3.0 in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from idasen-controller) (20.3.0)
Requirement already satisfied: incremental==17.5.0 in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from idasen-controller) (17.5.0)
Collecting Twisted==20.3.0
  Using cached Twisted-20.3.0.tar.bz2 (3.1 MB)
Requirement already satisfied: idna==3.1 in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from idasen-controller) (3.1)
Collecting txdbus==1.1.2
  Using cached txdbus-1.1.2.tar.gz (61 kB)
Requirement already satisfied: zope.interface==5.2.0 in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from idasen-controller) (5.2.0)
Requirement already satisfied: PyHamcrest==2.0.2 in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from idasen-controller) (2.0.2)
Requirement already satisfied: six==1.15.0 in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from idasen-controller) (1.15.0)
Collecting bleak==0.10.0
  Using cached bleak-0.10.0-py2.py3-none-any.whl (103 kB)
Collecting PyYAML==5.4.1
  Using cached PyYAML-5.4.1-cp39-cp39-win_amd64.whl (213 kB)
Requirement already satisfied: constantly==15.1.0 in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from idasen-controller) (15.1.0)
Collecting appdirs==1.4.4
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Requirement already satisfied: hyperlink==21.0.0 in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from idasen-controller) (21.0.0)
Requirement already satisfied: Automat==20.2.0 in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from idasen-controller) (20.2.0)
Collecting pythonnet
  Using cached pythonnet-2.5.2.tar.gz (1.9 MB)
Requirement already satisfied: setuptools in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from zope.interface==5.2.0->idasen-controller) (56.0.0)
Requirement already satisfied: pycparser in c:\users\eugene-pc\appdata\local\programs\python\python39\lib\site-packages (from pythonnet->bleak==0.10.0->idasen-controller) (2.20)
Using legacy 'setup.py install' for Twisted, since package 'wheel' is not installed.
Using legacy 'setup.py install' for txdbus, since package 'wheel' is not installed.
Using legacy 'setup.py install' for pythonnet, since package 'wheel' is not installed.
Installing collected packages: Twisted, pythonnet, txdbus, PyYAML, bleak, appdirs, idasen-controller
    Running setup.py install for Twisted ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\eugene-pc\appdata\local\programs\python\python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Eugene-PC\\AppData\\Local\\Temp\\pip-install-_clmldcg\\twisted_7ec3ca1963ec436fa5d8a8eaf5e52e0f\\setup.py'"'"'; __file__='"'"'C:\\Users\\Eugene-PC\\AppData\\Local\\Temp\\pip-install-_clmldcg\\twisted_7ec3ca1963ec436fa5d8a8eaf5e52e0f\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Eugene-PC\AppData\Local\Temp\pip-record-i117nomr\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\eugene-pc\appdata\local\programs\python\python39\Include\Twisted'
         cwd: C:\Users\Eugene-PC\AppData\Local\Temp\pip-install-_clmldcg\twisted_7ec3ca1963ec436fa5d8a8eaf5e52e0f\
    Complete output (984 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.9
    creating build\lib.win-amd64-3.9\twisted
    copying src\twisted\copyright.py -> build\lib.win-amd64-3.9\twisted
    copying src\twisted\plugin.py -> build\lib.win-amd64-3.9\twisted
    copying src\twisted\_version.py -> build\lib.win-amd64-3.9\twisted
    copying src\twisted\__init__.py -> build\lib.win-amd64-3.9\twisted
    copying src\twisted\__main__.py -> build\lib.win-amd64-3.9\twisted
    creating build\lib.win-amd64-3.9\twisted\application
    copying src\twisted\application\app.py -> build\lib.win-amd64-3.9\twisted\application
    copying src\twisted\application\internet.py -> build\lib.win-amd64-3.9\twisted\application
    copying src\twisted\application\reactors.py -> build\lib.win-amd64-3.9\twisted\application
    copying src\twisted\application\service.py -> build\lib.win-amd64-3.9\twisted\application
    copying src\twisted\application\strports.py -> build\lib.win-amd64-3.9\twisted\application
    copying src\twisted\application\__init__.py -> build\lib.win-amd64-3.9\twisted\application
    creating build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\avatar.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\checkers.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\endpoints.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\error.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\interfaces.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\ls.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\manhole.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\manhole_ssh.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\manhole_tap.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\mixin.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\recvline.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\stdio.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\tap.py -> build\lib.win-amd64-3.9\twisted\conch
    copying src\twisted\conch\telnet.py -> build\lib.win-amd64-3.9\twisted\conch
[... 926 lines cut ...]
    creating build\temp.win-amd64-3.9\Release\src\twisted\internet\iocpreactor\iocpsupport
    D:\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DWIN32=1 -Ic:\users\eugene-pc\appdata\local\programs\python\python39\include -Ic:\users\eugene-pc\appdata\local\programs\python\python39\include -ID:\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\include -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcsrc/twisted/internet/iocpreactor/iocpsupport/iocpsupport.c /Fobuild\temp.win-amd64-3.9\Release\src/twisted/internet/iocpreactor/iocpsupport/iocpsupport.obj
    iocpsupport.c
    src/twisted/internet/iocpreactor/iocpsupport/iocpsupport.c(2229): warning C4047: '=': '__pyx_t_11iocpsupport_HANDLE' differs in levels of indirection from 'HANDLE'
    src/twisted/internet/iocpreactor/iocpsupport/iocpsupport.c(2377): warning C4022: 'CreateIoCompletionPort': pointer mismatch for actual parameter 1
    src/twisted/internet/iocpreactor/iocpsupport/iocpsupport.c(2377): warning C4022: 'CreateIoCompletionPort': pointer mismatch for actual parameter 2
    src/twisted/internet/iocpreactor/iocpsupport/iocpsupport.c(2377): warning C4047: '=': '__pyx_t_11iocpsupport_HANDLE' differs in levels of indirection from 'HANDLE'
    src/twisted/internet/iocpreactor/iocpsupport/iocpsupport.c(2492): warning C4022: 'GetQueuedCompletionStatus': pointer mismatch for actual parameter 1
    src/twisted/internet/iocpreactor/iocpsupport/iocpsupport.c(2838): warning C4022: 'PostQueuedCompletionStatus': pointer mismatch for actual parameter 1
    src/twisted/internet/iocpreactor/iocpsupport/iocpsupport.c(2949): warning C4022: 'CloseHandle': pointer mismatch for actual parameter 1
    src/twisted/internet/iocpreactor/iocpsupport/iocpsupport.c(3292): warning C4996: 'PyObject_AsReadBuffer': deprecated in 3.0
[... cut ...] "', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Eugene-PC\AppData\Local\Temp\pip-record-i117nomr\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\eugene-pc\appdata\local\programs\python\python39\Include\Twisted' Check the logs for full command output.

rhyst, 2021-05-23:
I suggest you search for the error that appears towards the end:

```
error: command 'D:\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\bin\HostX86\x64\cl.exe' failed with exit code 2
```

Here is a stack overflow page with some things to try: https://stackoverflow.com/questions/50494411/error-command-c-visual-studio-2017-community-vc-tools-msvc-14-14-2642

It seems like it could be caused by many things, and is unlikely to be specific to `idasen-controller`.  

If you solve it then let me know and I can add it to the troubleshooting section.

rhyst, 2021-05-23:
Another random thought is that it seems to be an error with `Twisted` which is often problematic. You could try installing the wheel from https://www.lfd.uci.edu/~gohlke/pythonlibs/#twisted and then retrying the pip install of `idasen-controller`.

rhyst, 2021-06-14:
Closing as I don't think this is specifically to do with idasen controller.

closed 2021-06-14 by rhyst