rhyst/linak-controller #22: "abort"

rcambrj, 2021-08-01:
I'm struggling to run this, perhaps you can help me?

```
pip3 install idasen-controller
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Requirement already satisfied: idasen-controller in /usr/local/lib/python3.9/site-packages (1.0.5)
Requirement already satisfied: attrs==20.3.0 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (20.3.0)
Requirement already satisfied: idna==3.1 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (3.1)
Requirement already satisfied: Automat==20.2.0 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (20.2.0)
Requirement already satisfied: Twisted==20.3.0 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (20.3.0)
Requirement already satisfied: PyHamcrest==2.0.2 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (2.0.2)
Requirement already satisfied: txdbus==1.1.2 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (1.1.2)
Requirement already satisfied: zope.interface==5.2.0 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (5.2.0)
Requirement already satisfied: hyperlink==21.0.0 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (21.0.0)
Requirement already satisfied: six==1.15.0 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (1.15.0)
Requirement already satisfied: appdirs==1.4.4 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (1.4.4)
Requirement already satisfied: incremental==17.5.0 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (17.5.0)
Requirement already satisfied: constantly==15.1.0 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (15.1.0)
Requirement already satisfied: bleak==0.10.0 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (0.10.0)
Requirement already satisfied: PyYAML==5.4.1 in /usr/local/lib/python3.9/site-packages (from idasen-controller) (5.4.1)
Requirement already satisfied: pyobjc-core in /usr/local/lib/python3.9/site-packages (from bleak==0.10.0->idasen-controller) (7.3)
Requirement already satisfied: pyobjc-framework-CoreBluetooth in /usr/local/lib/python3.9/site-packages (from bleak==0.10.0->idasen-controller) (7.3)
Requirement already satisfied: pyobjc-framework-libdispatch in /usr/local/lib/python3.9/site-packages (from bleak==0.10.0->idasen-controller) (7.3)
Requirement already satisfied: setuptools in /usr/local/lib/python3.9/site-packages (from zope.interface==5.2.0->idasen-controller) (57.0.0)
Requirement already satisfied: pyobjc-framework-Cocoa>=7.3 in /usr/local/lib/python3.9/site-packages (from pyobjc-framework-CoreBluetooth->bleak==0.10.0->idasen-controller) (7.3)
```

```
idasen-controller
[1]    4255 abort      idasen-controller
```

```
python3 idasen_controller/main.py
[1]    4229 abort      python3 idasen_controller/main.py
```

I'm not super familiar with python, but I've managed to get this stack trace, maybe it's helpful?

```
python3 -X dev idasen_controller/main.py
Fatal Python error: Aborted

Thread 0x00000001142f5e00 (most recent call first):
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 316 in wait
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 574 in wait
  File "/usr/local/lib/python3.9/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 91 in init
  File "/usr/local/lib/python3.9/site-packages/bleak/backends/corebluetooth/scanner.py", line 37 in __init__
  File "~/projects/idasen/idasen-controller/idasen_controller/main.py", line 279 in scan
  File "~/projects/idasen/idasen-controller/idasen_controller/main.py", line 301 in connect
  File "~/projects/idasen/idasen-controller/idasen_controller/main.py", line 401 in main
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/events.py", line 80 in _run
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 1882 in _run_once
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 596 in run_forever
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 629 in run_until_complete
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/runners.py", line 44 in run
  File "~/projects/idasen/idasen-controller/idasen_controller/main.py", line 418 in init
  File "~/projects/idasen/idasen-controller/idasen_controller/main.py", line 423 in <module>
[1]    4275 killed     python3 -X dev idasen_controller/main.py
```

and line 316 of `threading.py` looks like this:

```
        if not self._is_owned():
            raise RuntimeError("cannot wait on un-acquired lock")
        waiter = _allocate_lock()
        waiter.acquire()
        self._waiters.append(waiter)
        saved_state = self._release_save()
        gotit = False
        try:    # restore state no matter what (e.g., KeyboardInterrupt)
            if timeout is None:
                waiter.acquire()
                gotit = True
            else:
                if timeout > 0:
                    gotit = waiter.acquire(True, timeout) <==== 316
                else:
                    gotit = waiter.acquire(False)
            return gotit
        finally:
            self._acquire_restore(saved_state)
            if not gotit:
                try:
                    self._waiters.remove(waiter)
                except ValueError:
                    pass
```

rhyst, 2021-08-02:
The stack trace doesn't mean much to me unfortunately. Assuming this happened during install, can you try running without installing from it pip? 

https://github.com/rhyst/idasen-controller#development

That will at least identify if its a problem with one of the dependencies installing or if its directly idasen-controller.

rcambrj, 2021-08-02:
@rhyst thanks for the quick reply, I don't think my last comment was very clear 😳 

when running the pip version:
```
idasen-controller
[1]    4255 abort      idasen-controller
```

when running from the cloned git repository
```
python3 idasen_controller/main.py
[1]    4229 abort      python3 idasen_controller/main.py
```

rhyst, 2021-08-02:
Ah, okay. Looks like something is going wrong in the `bleak` library scanning call. 

Can you try just bumping the bleak version in `requirements.txt`, reinstalling, and re-running? Try the latest `bleak==0.12.1`

rcambrj, 2021-08-02:
That appears to have no impact.

rhyst, 2021-08-02:
It would be worth finding out if `bleak` works at all for you. Can you try making and running a test script:
```
import asyncio
from bleak import BleakScanner

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

rcambrj, 2021-08-02:
I'm not sure how to get a better stack trace or more useful error at this point... `abort` is pretty vague :D 

I just tried `BLEAK_LOGGING=1` as recommended on the [Bleak troubleshooting page](https://bleak.readthedocs.io/en/latest/troubleshooting.html) but that doesn't seem to change the output.

rhyst, 2021-08-02:
Ah, well I don't mean to fob you off but I think you should raise an issue on the `bleak` repo 🙈 

Sorry I'm not more helpful!

rcambrj, 2021-08-02:
Good point. I'll open an issue/PR with guidance once I get it solved. Thanks for your help!

rcambrj, 2021-08-03:
Ugh, this was it! https://github.com/hbldh/bleak/issues/438#issuecomment-787125189

I'll make a PR for a troubleshooting section in the README this evening :)

rhyst, 2021-08-03:
Merged

closed 2021-08-03 by rhyst