rhyst/linak-controller #23: "Scanning - Desk Found" then quit

rcambrj, 2021-08-03:
I'm getting this problem when using `1.0.5`:
```
$ idasen-controller
Scanning - Desk Found
```

it seems to be fixed in `1.0.6` but that isn't published to PyPi, please could you publish it?

```
$ gco a7794fdbe19471b892341cd746696c16f5c6f697
Note: switching to 'a7794fdbe19471b892341cd746696c16f5c6f697'.
HEAD is now at a7794fd Bump version to 1.0.5
$ python3 idasen_controller/main.py
Scanning - Desk Found
$ gco master
Previous HEAD position was a7794fd Bump version to 1.0.5
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
$ python3 idasen_controller/main.py
Scanning - Desk Found
Connected 244C39A2-8606-4F3E-89C0-9076694D765F
Height: 1255mm
Disconnected
```

rhyst, 2021-08-03:
Published!

rcambrj, 2021-08-03:
Thanks for the quick turnaround 💯

closed 2021-08-03 by rcambrj

tcl89, 2021-11-09:
I'm having this issue in version 1.0.6 - any help would be appreciated!
Using Python 3.8.5

Xlinx64, 2021-11-09:
Hey @tcl89 
I saw this issue on linux when the desk is not paired. You can try to pair in with bluetoothctl.

tcl89, 2021-11-09:
Amazing - thank you. 
I'm on Windows - but pairing was the issue here too. Paired via Bluetooth LE Explorer, now it's working!