rhyst/linak-controller #26: Cannot install

MagnusModig, 2021-11-08:
Hello, I have Python 3.10 installed on a Windows 11 pc but cannot get the installer to complete. I get a lot of errors and one of them is 
error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/

Bu I cannot install hat version since there is a newer installed

I also get this error..

 ERROR: Command errored out with exit status 1:
     command: 'C:\Users\UserName\AppData\Local\Programs\Python\Python310\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\UserName\\AppData\\Local\\Temp\\pip-install-lvysg19b\\twisted_5d5952f5542d4b7fb0e29c24c66611d8\\setup.py'"'"'; __file__='"'"'C:\\Users\\UserName\\AppData\\Local\\Temp\\pip-install-lvysg19b\\twisted_5d5952f5542d4b7fb0e29c24c66611d8\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\UserName\AppData\Local\Temp\pip-record-dff81iw4\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\UserName\AppData\Local\Programs\Python\Python310\Include\Twisted'
         cwd: C:\Users\UserName\AppData\Local\Temp\pip-install-lvysg19b\twisted_5d5952f5542d4b7fb0e29c24c66611d8\

All this when running installer..

pip3 install idasen-controller

Hope you can help or have some ideas...

rhyst, 2021-11-08:
Looks like twisted is failing to install maybe, try installing it from here: https://www.lfd.uci.edu/~gohlke/pythonlibs/#twisted 

Find the appropriate `whl` file for your python version and architecture. I.e for python 3.9 and a 64 bit processor download `Twisted‑20.3.0‑cp39‑cp39‑win_amd64.whl`

And then you can run `pip install Twisted‑20.3.0‑cp39‑cp39‑win_amd64.whl` and then try installing idasen controller again.

MagnusModig, 2021-11-08:
Thanks, I tried the following..pip install twisted_iocpsupport-1.0.2-cp310-cp310-win_amd64.whl but I get this error..

ERROR: Could not install packages due to an OSError: [Errno 2] No such file or directory: 'C:\\Windows\\system32\\twisted_iocpsupport-1.0.2-cp310-cp310-win_amd64.whl'. Maybe I'm doing things wrong here....?

rhyst, 2021-11-08:
Maybe try `pip install Twisted[windows_platform]`

rhyst, 2021-11-08:
Also on that linked page is a whl called `twisted_iocpsupport` which is what your error says is missing so you could try installing that first.

rhyst, 2021-11-08:
Oh, are you in the same directory as the whl file as well when you run the `pip install` command?

MagnusModig, 2021-11-08:
pip install Twisted[windows_platform] worked but when I ran pip3 install idasen-controller

I got this error
ERROR: Command errored out with exit status 1:
     command: 'c:\users\magnus\appdata\local\programs\python\python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-jpla13ql\\twisted_bce2f9bf1c804da5b4871098c02af42c\\setup.py'"'"'; __file__='"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-jpla13ql\\twisted_bce2f9bf1c804da5b4871098c02af42c\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Magnus\AppData\Local\Temp\pip-record-ghi40ifx\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\magnus\appdata\local\programs\python\python39\Include\Twisted'
         cwd: C:\Users\Magnus\AppData\Local\Temp\pip-install-jpla13ql\twisted_bce2f9bf1c804da5b4871098c02af42c\

ERROR: Command errored out with exit status 1: 'c:\users\magnus\appdata\local\programs\python\python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-jpla13ql\\twisted_bce2f9bf1c804da5b4871098c02af42c\\setup.py'"'"'; __file__='"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-jpla13ql\\twisted_bce2f9bf1c804da5b4871098c02af42c\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Magnus\AppData\Local\Temp\pip-record-ghi40ifx\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\magnus\appdata\local\programs\python\python39\Include\Twisted' Check the logs for full command output.

MagnusModig, 2021-11-08:
> 
> 
> Oh, are you in the same directory as the whl file as well when you run the `pip install` command?

MagnusModig, 2021-11-08:
Sorry, I did not mean to close this issue... I still cannot install

rhyst, 2021-11-08:
You need to get `twisted` installed somehow and installing the `.whl` works for me on windows. Not sure why the `[windows_platform]` version does not work.

MagnusModig, 2021-11-08:
> [windows_platform]

I did get the twisted installed successfully using [windows_platform], but that does not help, I still get the errors I describe above

rhyst, 2021-11-08:
Please try to get the version from here https://www.lfd.uci.edu/~gohlke/pythonlibs/#twisted installed. That's the one that works for me. I do not know why `Twisted[windows_platform]` does not work.

MagnusModig, 2021-11-08:
> Twisted[windows_platform]

Please note that it does not work to install from https://www.lfd.uci.edu/~gohlke/pythonlibs/#twisted the only thing that works is installing by Twisted[windows_platform] but that does not make me install idasen

rhyst, 2021-11-08:
It looks like you downloaded the wrong file and also did not run the command in the correct directory.

Download `Twisted‑20.3.0‑cp39‑cp39‑win_amd64.whl` and then run `pip install Twisted‑20.3.0‑cp39‑cp39‑win_amd64.whl` in the correct directory. 

Note you can use three backticks (```) before and after logs to make them easier to read 

\`\`\`
some logs
\`\`\`

will look like
```
some logs
```

MagnusModig, 2021-11-08:
If I do that I get these errors...

ERROR: Command errored out with exit status 1:
     command: 'c:\users\magnus\appdata\local\programs\python\python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-e576tpjp\\pythonnet_cb8fcfd8902344c3aefc80c699feb851\\setup.py'"'"'; __file__='"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-e576tpjp\\pythonnet_cb8fcfd8902344c3aefc80c699feb851\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Magnus\AppData\Local\Temp\pip-record-54wqvrux\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\magnus\appdata\local\programs\python\python39\Include\pythonnet'
         cwd: C:\Users\Magnus\AppData\Local\Temp\pip-install-e576tpjp\pythonnet_cb8fcfd8902344c3aefc80c699feb851\
    Complete output (6 lines):
    usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: setup.py --help [cmd1 cmd2 ...]
       or: setup.py --help-commands
       or: setup.py cmd --help

    error: option --single-version-externally-managed not recognized
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\magnus\appdata\local\programs\python\python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-e576tpjp\\pythonnet_cb8fcfd8902344c3aefc80c699feb851\\setup.py'"'"'; __file__='"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-e576tpjp\\pythonnet_cb8fcfd8902344c3aefc80c699feb851\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Magnus\AppData\Local\Temp\pip-record-54wqvrux\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\magnus\appdata\local\programs\python\python39\Include\pythonnet' Check the logs for full command output.

rhyst, 2021-11-08:
I would try searching for and solving that error `error: option --single-version-externally-managed not recognized`

MagnusModig, 2021-11-21:
I tried again today, got these errors....

  ERROR: Command errored out with exit status 1:
   command: 'c:\users\magnus\appdata\local\programs\python\python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-8o_77a89\\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\\setup.py'"'"'; __file__='"'"'C:\\Users\\Magnus\\AppData\\Local\\Temp\\pip-install-8o_77a89\\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\Magnus\AppData\Local\Temp\pip-wheel-22x7pce1'
       cwd: C:\Users\Magnus\AppData\Local\Temp\pip-install-8o_77a89\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\
  Complete output (43 lines):
  running bdist_wheel
  running build
  running build_ext
  Checking for updates from https://www.nuget.org/api/v2/.
  Currently running NuGet.exe 4.1.0.
  Updating NuGet.exe to 5.11.0.
  Update successful.
  Cannot find the specified version of msbuild: '14'
  MSBuild auto-detection: using msbuild version '17.0.0.52104' from 'C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin'.
  C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Microsoft.CSharp.CurrentVersion.targets(318,5): error MSB4019: The imported project "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" was not found. Confirm that the expression in the Import declaration "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" is correct, and that the file exists on disk. [C:\Users\Magnus\AppData\Local\Temp\pip-install-8o_77a89\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\src\runtime\Python.Runtime.csproj]
  C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Microsoft.CSharp.CurrentVersion.targets(318,5): error MSB4019: The imported project "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" was not found. Confirm that the expression in the Import declaration "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" is correct, and that the file exists on disk. [C:\Users\Magnus\AppData\Local\Temp\pip-install-8o_77a89\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\src\testing\Python.Test.csproj]
  C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Microsoft.CSharp.CurrentVersion.targets(318,5): error MSB4019: The imported project "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" was not found. Confirm that the expression in the Import declaration "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" is correct, and that the file exists on disk. [C:\Users\Magnus\AppData\Local\Temp\pip-install-8o_77a89\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\src\embed_tests\Python.EmbeddingTest.csproj]
  C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Microsoft.CSharp.CurrentVersion.targets(318,5): error MSB4019: The imported project "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" was not found. Confirm that the expression in the Import declaration "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" is correct, and that the file exists on disk. [C:\Users\Magnus\AppData\Local\Temp\pip-install-8o_77a89\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\src\console\Console.csproj]
  C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Microsoft.CSharp.CurrentVersion.targets(318,5): error MSB4019: The imported project "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" was not found. Confirm that the expression in the Import declaration "C:\Program Files\Microsoft Visual Studio\2022\TeamExplorer\MSBuild\Current\bin\Roslyn\Microsoft.CSharp.Core.targets" is correct, and that the file exists on disk. [C:\Users\Magnus\AppData\Local\Temp\pip-install-8o_77a89\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\src\clrmodule\clrmodule.csproj]

  WARNING: Error reading msbuild project information, ensure that your input solution or project file is valid. NETCore and UAP projects will be skipped, only packages.config files will be restored.
  Restoring NuGet package NUnit.3.12.0.
  Restoring NuGet package System.ValueTuple.4.5.0.
  Restoring NuGet package NUnit.ConsoleRunner.3.11.1.
  Restoring NuGet package UnmanagedExports.1.2.7.
  Adding package 'NUnit.3.12.0' to folder 'C:\Users\Magnus\AppData\Local\Temp\pip-install-8o_77a89\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\packages'
  Adding package 'UnmanagedExports.1.2.7' to folder 'C:\Users\Magnus\AppData\Local\Temp\pip-install-8o_77a89\pythonnet_a4a7f4e528c84bc8a9bebc576a5ec018\packages'
[... cut ...] n'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Magnus\AppData\Local\Temp\pip-record-ks4jie40\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\magnus\appdata\local\programs\python\python39\Include\pythonnet' Check the logs for full command output.

MagnusModig, 2021-11-21:
I also tried every suggestion in this thread [https://stackoverflow.com/questions/14296531/what-does-error-option-single-version-externally-managed-not-recognized-ind](url), but that doesnt help either

rhyst, 2021-11-21:
1. What command did you run to get that error? It's different to the previous error so have you done anything differently? It's helpful to include more context than just the console log.
2. Which version of twisted have you got installed? If it is not the one from [here](https://www.lfd.uci.edu/~gohlke/pythonlibs/#twisted) then I suggest trying again to install that version as it is the one that works for me on Windows with Python 3.9
3. Please try to format your logs with the ``` as I explained above, it makes them much easier to read.

MagnusModig, 2021-11-21:
I ran pip3 install idasen-controller, I'll give the twisted another shot and get back...

rhyst, 2022-01-07:
I realised that quite a lot of those dependencies were unneeded. I just pushed 1.0.7 which does not require twisted, and it seems to install more easily on windows.

MagnusModig, 2022-01-08:
Thanks the installation now went ok. But when I run idasen-controller command I get this error...

Scanning - Desk Found
Traceback (most recent call last):
  File "c:\users\magnus\appdata\local\programs\python\python39\lib\site-packages\idasen_controller\main.py", line 401, in main
    client = await connect()
  File "c:\users\magnus\appdata\local\programs\python\python39\lib\site-packages\idasen_controller\main.py", line 311, in connect
    await client.connect(timeout=config['connection_timeout'])
  File "c:\users\magnus\appdata\local\programs\python\python39\lib\site-packages\bleak\backends\winrt\client.py", line 257, in connect
    await self.get_services()
  File "c:\users\magnus\appdata\local\programs\python\python39\lib\site-packages\bleak\backends\winrt\client.py", line 426, in get_services
    await self._requester.get_gatt_services_async(
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\magnus\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\magnus\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Magnus\AppData\Local\Programs\Python\Python39\Scripts\idasen-controller.exe\__main__.py", line 7, in <module>
  File "c:\users\magnus\appdata\local\programs\python\python39\lib\site-packages\idasen_controller\main.py", line 418, in init
    asyncio.run(main())
  File "c:\users\magnus\appdata\local\programs\python\python39\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "c:\users\magnus\appdata\local\programs\python\python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
asyncio.exceptions.CancelledError

I have set the MAC adress of my desk in the yaml file...

rhyst, 2022-01-08:
I think that's progress 😆 

It looks like it might be a problem with the Bleak library, and there seem to be some issues open which sounds vaguelllyyyy related: https://github.com/hbldh/bleak/issues?q=is%3Aissue+is%3Aopen+label%3A%22Backend%3A+WinRT%22

You could try increasing the `connection_timeout` ?

MagnusModig, 2022-01-11:
increase timeout does not help. Running the script you provided give me this error..

D:\>bleak.py
Traceback (most recent call last):
  File "D:\bleak.py", line 6, in <module>
    from bleak import BleakClient, BleakScanner
  File "D:\bleak.py", line 6, in <module>
    from bleak import BleakClient, BleakScanner
ImportError: cannot import name 'BleakClient' from partially initialized module 'bleak' (most likely due to a circular import) (D:\bleak.py)

rhyst, 2022-01-11:
A google search suggests that might be because you've named it `bleak.py`. Try renaming it to `test.py` or something. 

I think its attempting to import `BleakClient` and `BleakScanner` from `bleak.py` rather than the `bleak` library.

MagnusModig, 2022-01-11:
Sorry still same error...

Traceback (most recent call last):
  File "D:\tests.py", line 6, in <module>
    from bleak import BleakClient, BleakScanner
  File "D:\bleak.py", line 6, in <module>
    from bleak import BleakClient, BleakScanner
ImportError: cannot import name 'BleakClient' from partially initialized module 'bleak' (most likely due to a circular import) (D:\bleak.py)

rhyst, 2022-01-11:
That error still makes it look like you've got a file called `bleak.py`. Delete that as its still trying to import from there.

MagnusModig, 2022-01-11:
ahh, that made a difference. Now the bleak script works.

rhyst, 2022-01-11:
What's the output of the script? I'm not ent

rhyst, 2022-01-11:
Oh also can you confirm that you've already paired the desk in windows?

MagnusModig, 2022-01-11:
I can confirm that the desk is paired with windows. Here is the output of the command above...

2022-01-11 18:39:57,190 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 18:39:57,191 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 18:39:57,630 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 18:39:57,631 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 18:39:57,728 bleak.backends.winrt.scanner DEBUG: Received E4:93:33:9A:6B:F3: P mesh.
2022-01-11 18:39:57,731 bleak.backends.winrt.scanner DEBUG: Received E4:93:33:9A:6B:F3: Unknown.
2022-01-11 18:39:57,735 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: Unknown.
2022-01-11 18:39:57,739 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: [TV] Samsung QN90AA 55 TV.
2022-01-11 18:39:57,843 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 18:39:57,845 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 18:39:57,851 bleak.backends.winrt.scanner DEBUG: Received 9C:20:7B:94:AE:A8: Unknown.
2022-01-11 18:39:57,852 bleak.backends.winrt.scanner DEBUG: Received 9C:20:7B:94:AE:A8: Unknown.
2022-01-11 18:39:58,213 bleak.backends.winrt.scanner DEBUG: Received 9C:20:7B:94:AE:A8: Unknown.
2022-01-11 18:39:58,214 bleak.backends.winrt.scanner DEBUG: Received 9C:20:7B:94:AE:A8: Unknown.
2022-01-11 18:39:58,326 bleak.backends.winrt.scanner DEBUG: Received 6D:30:29:37:1D:4E: Unknown.
2022-01-11 18:39:58,328 bleak.backends.winrt.scanner DEBUG: Received 6D:30:29:37:1D:4E: SHIELD.
2022-01-11 18:39:58,430 bleak.backends.winrt.scanner DEBUG: Received 65:FB:54:07:84:75: Unknown.
2022-01-11 18:39:58,550 bleak.backends.winrt.scanner DEBUG: Received 46:7C:04:89:ED:1C: Unknown.
2022-01-11 18:39:58,551 bleak.backends.winrt.scanner DEBUG: Received 46:7C:04:89:ED:1C: Unknown.
2022-01-11 18:39:58,682 bleak.backends.winrt.scanner DEBUG: Received DE:EF:BA:DD:3E:B6: Unknown.
2022-01-11 18:39:58,689 bleak.backends.winrt.scanner DEBUG: Received 65:FB:54:07:84:75: Unknown.
2022-01-11 18:39:58,692 bleak.backends.winrt.scanner DEBUG: Received 65:FB:54:07:84:75: Unknown.
2022-01-11 18:39:58,785 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: Unknown.
2022-01-11 18:39:58,788 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: [TV] Samsung QN90AA 55 TV.
2022-01-11 18:39:58,901 bleak.backends.winrt.scanner DEBUG: Received 46:7C:04:89:ED:1C: Unknown.
2022-01-11 18:39:58,905 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 18:39:58,907 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 18:39:59,023 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 18:39:59,025 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 18:39:59,371 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 18:39:59,373 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 18:39:59,391 bleak.backends.winrt.scanner DEBUG: Received 7E:E0:8F:D4:28:E2: Unknown.
2022-01-11 18:39:59,394 bleak.backends.winrt.scanner DEBUG: Received 7E:E0:8F:D4:28:E2: Unknown.
2022-01-11 18:39:59,487 bleak.backends.winrt.scanner DEBUG: Received DE:EF:BA:DD:3E:B6: Unknown.
2022-01-11 18:39:59,494 bleak.backends.winrt.scanner DEBUG: Received DE:EF:BA:DD:3E:B6: Unknown.
2022-01-11 18:39:59,499 bleak.backends.winrt.scanner DEBUG: Received 46:7C:04:89:ED:1C: Unknown.
2022-01-11 18:39:59,501 bleak.backends.winrt.scanner DEBUG: Received DE:EF:BA:DD:3E:B6: Unknown.
2022-01-11 18:39:59,506 bleak.backends.winrt.scanner DEBUG: Received 46:7C:04:89:ED:1C: Unknown.
2022-01-11 18:39:59,507 bleak.backends.winrt.scanner DEBUG: Received 46:7C:04:89:ED:1C: Unknown.
2022-01-11 18:39:59,539 bleak.backends.winrt.scanner DEBUG: Received FA:86:90:0F:AF:A7: Unknown.
2022-01-11 18:39:59,726 bleak.backends.winrt.scanner DEBUG: Received 65:FB:54:07:84:75: Unknown.
2022-01-11 18:39:59,844 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: Unknown.
2022-01-11 18:39:59,847 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: [TV] Samsung QN90AA 55 TV.
2022-01-11 18:40:00,322 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 18:40:00,325 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 18:40:00,427 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 18:40:00,430 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 18:40:00,436 bleak.backends.winrt.scanner DEBUG: Received 7E:E0:8F:D4:28:E2: Unknown.
2022-01-11 18:40:00,439 bleak.backends.winrt.scanner DEBUG: Received 7E:E0:8F:D4:28:E2: Unknown.
2022-01-11 18:40:00,549 bleak.backends.winrt.scanner DEBUG: Received E4:93:33:9A:6B:F3: P mesh.
2022-01-11 18:40:00,782 bleak.backends.winrt.scanner DEBUG: Received 67:07:57:58:34:3F: Unknown.
2022-01-11 18:40:00,785 bleak.backends.winrt.scanner DEBUG: Received 67:07:57:58:34:3F: Unknown.
2022-01-11 18:40:00,822 bleak.backends.winrt.scanner DEBUG: Received 6D:30:29:37:1D:4E: Unknown.
2022-01-11 18:40:00,824 bleak.backends.winrt.scanner DEBUG: Received 6D:30:29:37:1D:4E: SHIELD.
2022-01-11 18:40:00,905 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: Unknown.
[... cut ...]
  File "D:\test2.py", line 11, in run
    print(await client.connect(timeout=30))
  File "C:\Users\Magnus\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\backends\winrt\client.py", line 161, in connect
    device = await BleakScannerWinRT.find_device_by_address(
  File "C:\Users\Magnus\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\backends\scanner.py", line 203, in find_device_by_address
    device_identifier = device_identifier.lower()
AttributeError: 'NoneType' object has no attribute 'lower'

rhyst, 2022-01-11:
Did you update the mac address to your own? It looks like `address` is `None` somehow?

MagnusModig, 2022-01-11:
Sorry, I did not, but when I change it I get this result...

2022-01-11 18:45:39,602 bleak.backends.winrt.scanner DEBUG: Received E4:93:33:9A:6B:F3: P mesh.
2022-01-11 18:45:39,603 bleak.backends.winrt.scanner DEBUG: Received E4:93:33:9A:6B:F3: Unknown.
2022-01-11 18:45:39,607 bleak.backends.winrt.scanner DEBUG: Received 7E:E0:8F:D4:28:E2: Unknown.
2022-01-11 18:45:39,610 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 18:45:39,611 bleak.backends.winrt.scanner DEBUG: 3 devices found. Watcher status: 3.
2022-01-11 18:45:39,611 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ E7:36:0E:FA:D9:91
2022-01-11 18:45:40,071 bleak.backends.winrt.client DEBUG: _ConnectionStatusChanged_Handler: 1
2022-01-11 18:45:40,072 bleak.backends.winrt.client DEBUG: Get Services...
2022-01-11 18:45:43,671 bleak.backends.winrt.client INFO: Services resolved for BleakClientWinRT (E7:36:0E:FA:D9:91)

rhyst, 2022-01-11:
I would expect the final line to be `True` or `False`, is that all it printed?

rhyst, 2022-01-11:
Ah weird, its connecting okay and that's basically how it connects in `idasen-controller` as well. 

Can you modify one more time to be exactly how `idasen-controller` does it and post the results?

```
import os
os.environ["BLEAK_LOGGING"] = "1"
import asyncio
from bleak import BleakClient, BleakScanner

address = "E8:5B:5B:24:22:E4" # YOUR MAC ADDRESS

async def run(address):
    scanner = BleakScanner()
    found = None
    devices = await scanner.discover(timeout=10)
    for device in devices:
        if (device.address == address):
            print('found')
            found = device
    if found:
        client = BleakClient(device)
        print(await client.connect(timeout=30))
    else:
        print('not found')

asyncio.run(run(address))
```

I should probably update this to just use `find_device_by_address`...

MagnusModig, 2022-01-11:
I then get this...

2022-01-11 21:08:19,432 bleak.backends.winrt.scanner DEBUG: Received 4B:A2:B6:77:3D:23: Unknown.
2022-01-11 21:08:19,434 bleak.backends.winrt.scanner DEBUG: Received 4B:A2:B6:77:3D:23: Unknown.
2022-01-11 21:08:19,669 bleak.backends.winrt.scanner DEBUG: Received EE:85:9C:8C:54:BA: P mesh.
2022-01-11 21:08:19,670 bleak.backends.winrt.scanner DEBUG: Received EE:85:9C:8C:54:BA: Unknown.
2022-01-11 21:08:19,783 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 21:08:19,784 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 21:08:19,892 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 21:08:19,895 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 21:08:20,005 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 21:08:20,007 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 21:08:20,012 bleak.backends.winrt.scanner DEBUG: Received 6A:AB:2B:37:1E:0C: Unknown.
2022-01-11 21:08:20,014 bleak.backends.winrt.scanner DEBUG: Received 6A:AB:2B:37:1E:0C: SHIELD.
2022-01-11 21:08:20,145 bleak.backends.winrt.scanner DEBUG: Received 6E:0D:5C:1D:9E:CF: Unknown.
2022-01-11 21:08:20,592 bleak.backends.winrt.scanner DEBUG: Received 75:BA:20:C9:8E:20: Unknown.
2022-01-11 21:08:20,593 bleak.backends.winrt.scanner DEBUG: Received 75:BA:20:C9:8E:20: Unknown.
2022-01-11 21:08:20,615 bleak.backends.winrt.scanner DEBUG: Received EE:85:9C:8C:54:BA: P mesh.
2022-01-11 21:08:20,619 bleak.backends.winrt.scanner DEBUG: Received EE:85:9C:8C:54:BA: Unknown.
2022-01-11 21:08:20,726 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 21:08:20,727 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 21:08:20,729 bleak.backends.winrt.scanner DEBUG: Received 4B:A2:B6:77:3D:23: Unknown.
2022-01-11 21:08:20,730 bleak.backends.winrt.scanner DEBUG: Received 4B:A2:B6:77:3D:23: Unknown.
2022-01-11 21:08:20,833 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 21:08:20,841 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 21:08:20,849 bleak.backends.winrt.scanner DEBUG: Received 9C:20:7B:94:AE:A8: Unknown.
2022-01-11 21:08:20,857 bleak.backends.winrt.scanner DEBUG: Received 6A:AB:2B:37:1E:0C: Unknown.
2022-01-11 21:08:20,858 bleak.backends.winrt.scanner DEBUG: Received 6A:AB:2B:37:1E:0C: SHIELD.
2022-01-11 21:08:20,859 bleak.backends.winrt.scanner DEBUG: Received 7B:B0:35:63:CF:A6: Unknown.
2022-01-11 21:08:20,861 bleak.backends.winrt.scanner DEBUG: Received 7B:B0:35:63:CF:A6: Unknown.
2022-01-11 21:08:21,298 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: Unknown.
2022-01-11 21:08:21,299 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: [TV] Samsung QN90AA 55 TV.
2022-01-11 21:08:21,310 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: Unknown.
2022-01-11 21:08:21,313 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: [TV] Samsung QN90AA 55 TV.
2022-01-11 21:08:21,436 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 21:08:21,444 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 21:08:21,543 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 21:08:21,551 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 21:08:21,554 bleak.backends.winrt.scanner DEBUG: Received DF:2E:DC:34:9E:04: P mesh.
2022-01-11 21:08:21,555 bleak.backends.winrt.scanner DEBUG: Received DF:2E:DC:34:9E:04: Unknown.
2022-01-11 21:08:21,771 bleak.backends.winrt.scanner DEBUG: Received 4B:A2:B6:77:3D:23: Unknown.
2022-01-11 21:08:21,774 bleak.backends.winrt.scanner DEBUG: Received 4B:A2:B6:77:3D:23: Unknown.
2022-01-11 21:08:22,006 bleak.backends.winrt.scanner DEBUG: Received 7E:33:0B:FD:5A:D0: Unknown.
2022-01-11 21:08:22,007 bleak.backends.winrt.scanner DEBUG: Received 7E:33:0B:FD:5A:D0: Unknown.
2022-01-11 21:08:22,011 bleak.backends.winrt.scanner DEBUG: Received 65:35:DD:C9:D0:5F: Unknown.
2022-01-11 21:08:22,013 bleak.backends.winrt.scanner DEBUG: Received 65:35:DD:C9:D0:5F: Unknown.
2022-01-11 21:08:22,251 bleak.backends.winrt.scanner DEBUG: Received 7F:FC:02:B3:26:E6: Unknown.
2022-01-11 21:08:22,254 bleak.backends.winrt.scanner DEBUG: Received 7F:FC:02:B3:26:E6: Unknown.
2022-01-11 21:08:22,255 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 21:08:22,257 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 21:08:22,357 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: Unknown.
2022-01-11 21:08:22,358 bleak.backends.winrt.scanner DEBUG: Received 64:07:F6:98:1F:AD: [TV] Samsung QN90AA 55 TV.
2022-01-11 21:08:22,360 bleak.backends.winrt.scanner DEBUG: Received EE:85:9C:8C:54:BA: P mesh.
2022-01-11 21:08:22,361 bleak.backends.winrt.scanner DEBUG: Received EE:85:9C:8C:54:BA: Unknown.
2022-01-11 21:08:22,365 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
2022-01-11 21:08:22,368 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Desk 5697.
2022-01-11 21:08:22,475 bleak.backends.winrt.scanner DEBUG: Received E7:36:0E:FA:D9:91: Unknown.
[... cut ...]
  File "C:\Users\Magnus\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "D:\test3.py", line 18, in run
    print(await client.connect(timeout=30))
  File "C:\Users\Magnus\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\backends\winrt\client.py", line 249, in connect
    await asyncio.wait_for(event.wait(), timeout=timeout)
  File "C:\Users\Magnus\AppData\Local\Programs\Python\Python39\lib\asyncio\tasks.py", line 491, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

rhyst, 2022-01-11:
I just pushed that change. If you've cloned the repository then just pull master again and try running it. If you haven't got the repository I think you can install straight from github like this:

```
pip install git+https://github.com/rhyst/idasen-controller.git
```

rhyst, 2022-01-11:
Ah if you do install it from git you might need to uninstall it first because I haven't bumped the version:

```
pip uninstall idasen-controller
pip install git+https://github.com/rhyst/idasen-controller.git
```

MagnusModig, 2022-01-11:
Hmmm I tried this...

C:\Users\Magnus>pip install git+https://github.com/rhyst/idasen-controller.git
Collecting git+https://github.com/rhyst/idasen-controller.git
  Cloning https://github.com/rhyst/idasen-controller.git to c:\users\magnus\appdata\local\temp\pip-req-build-vyy4xu9v
  ERROR: Error [WinError 2] The system cannot find the file specified while executing command git version
ERROR: Cannot find command 'git' - do you have 'git' installed and in your PATH?

rhyst, 2022-01-11:
You could install git but I've pushed it as an prerelease version so this should work:

```
pip install idasen-controller==1.0.8rc1
```

MagnusModig, 2022-01-11:
I installed git and ran the command again and it installed ok. But when I run command idasen-controller it is stucked in "connecting", its the same after installing 1.0.8rc1 too....

rhyst, 2022-01-11:
I am not sure how to debug this any further without any error messages. You could clone the repo and fiddle with it yourself. Once you've cloned the repo you can just run:

```
python idasen_controller/main.py
```

From the repo directory and you can edit `main.py` to try to see where its getting stuck. 

Alternatively `bleak` has a new version with some changes to the windows backend which might help (but its got bugs so I'm going to wait for them to fix it before using it here).

MagnusModig, 2022-01-12:
Sorry to hear that, it was a great way to control the desk when it worked.

closed 2022-01-12 by rhyst