rhyst/linak-controller #28: Unhandled error

backslashV, 2021-12-10:
Device: raspberry pi zero
``bluetoothctl -v: 5.6.2

Error:
```
Unhandled Error
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/protocol.py", line 140, in dataReceived
    self.rawDBusMessageReceived(raw_msg)
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/protocol.py", line 266, in rawDBusMessageReceived
    self.signalReceived(m)
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/client.py", line 642, in signalReceived
    self.router.routeMessage(msig)
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/router.py", line 128, in routeMessage
    r.match(m)
--- <exception caught here> ---
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/router.py", line 67, in match
    self.callback(m)
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 867, in _properties_changed_callback
    message.path, message.body[1]
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 894, in args_parser
    return func(char_map.get(sender, sender), bytearray(data.get("Value")))
  File "/home/pi/Desktop/idasen-controller/idasen_controller/main.py", line 224, in _move_to
    except asyncio.exceptions.InvalidStateError:
builtins.AttributeError: module 'asyncio' has no attribute 'exceptions'

Task exception was never retrieved
future: <Task finished coro=<unsubscribe() done, defined at /home/pi/Desktop/idasen-controller/idasen_controller/main.py:187> exception=RemoteError('org.bluez.Error.Failed')>
Traceback (most recent call last):
  File "/home/pi/Desktop/idasen-controller/idasen_controller/main.py", line 189, in unsubscribe
    await client.stop_notify(uuid)
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 776, in stop_notify
    ).asFuture(asyncio.get_event_loop())
txdbus.error.RemoteError: org.bluez.Error.Failed: No notify session started
ConnectingConnected F2:7D:17:78:73:80
Height:  675mm
Final height:  811mm (Target: 1050mm)
DisconnectingDisconnected
```

Any ideas? I was getting the same error before updating bluetoothctl when it was at v5.5.

rhyst, 2021-12-11:
What version of Python are you using? It looks like maybe `asyncio.exceptions.InvalidStateError` only works in >3.8. I think I can change it to  `asyncio.InvalidStateError` which should work down to 3.6.

I'm not sure if that's the _actual_ error though or if its just erroring there as well.

backslashV, 2021-12-14:
I am running Python 3.7. I changed it to `asyncio.InvalidStateError` in my forked repo. Will report if I still see the issue.

backslashV, 2021-12-18:
So, I upgraded to Python 3.9. The unhandled error part went away, but I am still seeing the "Task exception was never retrieved error" from time to time:
[image]

rhyst, 2022-01-08:
I don't have a RPi to test on unfortunately. Does it still move the desk correctly despite the errors, or does it stop? If you were willing to add some error handling in the right place I'd happily accept a PR.

rhyst, 2022-01-21:
Closing as it seems like it works despite the error?

closed 2022-01-21 by rhyst