rhyst/linak-controller #31: stop command | homebridge plugin

vniehues, 2022-01-16:
Hey @rhyst 

After buying a new Ikea Idasen I immediately wanted to integrate it into HomeKit. 
I found your work an started integrating it into a homebridge plugin. 
I'm just spawning your controller as a child_process.

There is no real readme for now but I'll make sure to credit you.
Here is a link to the repo:
https://github.com/vniehues/linak-homebridge

The one thing that im missing from your script is a way to immediately stop any movement of the desk. 
When I kill the movement process, the desk keeps moving. 
Is there any way you could integrate something like that in your controller? Is that even possible? 

Thank you for your work! 
It is outstanding :) 

Cheers

rhyst, 2022-01-17:
How long does it continue moving for? It relies on sending movement commands repeatedly so if you stop the script then the desk should stop pretty shortly afterwards.

There is a `stop` function in the code which is used but its not currently hooked up to directly to a command line flag. That could be done, though I would expect that by the time the connection had been made to send the command that the desk would have stopped by itself anyway.

vniehues, 2022-01-18:
Yeah I think you're right. 

My specific use case was to cancel the movement and start another one (to a different position).
Sometimes it would not cancel fast enough and the desk would start jittering (idk if that the right word) in place. Neither moving up or down but trying to do both. I think that was an issue with _my_ code though.

Thank you anyways

closed 2022-01-18 by vniehues