rhyst/linak-controller #33: Unable to connect using MacOS Monterey

ash3rr, 2022-01-17:
I have created a config.yaml file in the library folder, but when I run the command:
idasen-controller --config "/Library/Application Support/idasen-controller/config.yaml"
I receive the error:
Scanning - Desk D8-7D-63-0D-91-32 Not Found

config.yaml looks like this:
mac_address: D8-7D-63-0D-91-32
stand_height: 1040
sit_height: 683
height_tolerance: 2.0
scan_timeout: 5
connection_timeout: 10
movement_timeout: 30
adapter_name: hci0
server_address": "127.0.0.1"
server_port: 9123

rhyst, 2022-01-17:
Someone noted that on on MacOS you cannot use the MAC address and instead you must use the UUID found with when running `idasen-controller --scan`. Have you tried that?

ash3rr, 2022-01-17:
Yes, I did. I've connected the desk using BlueUtility, and then tried to run the scan. 
The returned message is: Found 0 devices using hci0

rhyst, 2022-01-17:
Have you connected using the UUID rather than the mac address? The config you posted above is using the mac address.

ash3rr, 2022-01-18:
How do you propose I get the UUID of the desk without installing XCode? 
idasen-controller --scan does not return the UUID as listed in your documentation, rather only 'found 0 devices'.

rhyst, 2022-01-18:
I propose that as the owner of a Mac that you are in a better position than me to find that out 😆 

But more seriously I don't know why the `--scan` function doesn't return anything. You could try using the bleak library directly to see if that returns anything:

```
import asyncio
from bleak import BleakScanner

async def main():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

asyncio.run(main())
```

https://bleak.readthedocs.io/en/latest/scanning.html

vniehues, 2022-01-21:
I can confirm the issue. 
Using bluetillity I found my desks UUID and tried connecting to it using 
`Idasen-controller --mac-address <UUID>`
The script says it can't find a desk under that mac address. 
Using the real mac address doesn't work either. 

It works perfectly on windows & linux but not on macOS

rhyst, 2022-01-21:
Thanks for the info. Would you be able to run the code snippet I posted above? I want to know if this is a problem with the idasen controller code or if its also in the bleak library.

vniehues, 2022-01-21:
Yes I ran the code and got absolutely no output. 
After adding some print() statements I got to see my prints but no discovered devices.
I think the issue is with bleak and not your usage of it.

vniehues, 2022-01-21:
After some searching around on the bleak repo I found many (many!) comments about macOS 12. 
There are workaround but nothing really usable in the context of a CLI. 
I think you can close this because it is evident that the issue is with bleak and it won't be fixed in the near future. 

References:
[comment](https://github.com/hbldh/bleak/issues/635#issuecomment-988054876)
[issue](https://github.com/hbldh/bleak/issues/720)
[comment](https://github.com/hbldh/bleak/issues/720#issuecomment-1009198158)

rhyst, 2022-01-21:
Ah yeh I was just poking around those comments. It seems to be an issue with scanning though and the idasen controller code does a scan before connecting even if you provide a UUID/mac address. 

I can't remember why so maybe we can drop that and it will work. 

Could you try running something like this example code to connect to the desk without scanning https://github.com/hbldh/bleak/blob/develop/examples/get_services.py

vniehues, 2022-01-21:
I tried the code with the UUID & Mac address. Unfortunately neither worked.
I also tried it with the desk actively searching (holding the bt button for 3 seconds) and without.

rhyst, 2022-01-21:
That is a shame but I've just pushed a version without the preconnection scanning anyway. I think I assumed there was some benefit to already having the BLEDevice object but looking at the bleak code all it does is immediately extract the address string from it 🤦‍♂️

vniehues, 2022-01-21:
RC3 is looking really good! 
It doesn't solve this issues but it's overall a good performance improvement. 
Do you have a timeline for the full release?

rhyst, 2022-01-21:
Just released it as 1.0.8, I was mostly doing the rcs as I often break the packages when uploading 😆 

Still using this issue to track the mac os 12 bug. It seems like most people are raising that _scanning_ does not work, but I'm unsure if that also means that other people are having connection issues. I think that might be worth raising on the Bleak repo at least to ask for clarification.

vniehues, 2022-01-21:
Sounds good. I submitted a feedback to apple regarding the general issue that bluetooth over cli seems to get no permissions even when terminal is allowed bluetooth access.

Let's hope for the best!

dKemen, 2022-02-02:
Same issue here - neither using "get_services.py" with the UUID provided by Bluetility nor "idasen-controller --scan" works for me. I tried using "Remote Desk Control" from the Appstore and this app was able to connect.

Would be happy if we could get feedback from apple or Bleak for this.

vniehues, 2022-03-08:
Apple responded to my feedback. It's in German but basically they believe that issue to be fixed in **macOS 12.3 Beta 4**.

I currently have no way to check this because I need my Mac as stable as possible but if someone want's to try it out and give us feedback, that would be highly appreciated.

Copy of the mail in German:

> Hallo Vincent,
> 
> vielen Dank für deine Geduld und deinen Feedback-Bericht. Wir glauben, dass dieses Problem mit den Updates für macOS behoben wurde.
> 
> Bitte überprüfe das Problem mit macOS 12.3 Beta 4 und aktualisiere deinen Feedback-Bericht mit deinen Ergebnissen. Melde dich dazu bei Feedback-Assistent-Website an an oder verwende die Feedback-Assistent-App. Bitte schließe diesen Bericht, wenn sich das Problem erledigt hat, oder gib uns Bescheid, wenn es weiterhin ein Problem für dich ist.

vniehues, 2022-03-10:
Today I found the time to update my Mac to the latest beta. The version I'm now running it the RC1 of 12.3

It works! 💯 

Here's a quick screenshot of my terminal on 12.3:
[image]

"Searching" means that I pressed the BT button on the controller for 3 seconds (led flashes quickly).

rhyst, 2022-03-10:
Ah nice, thank you for confirming! I will update the readme and close this issue soon.

rhyst, 2022-03-11:
Updated README

closed 2022-03-11 by rhyst