rhyst/linak-controller #34: Configurable BaseHeight & MaxHeight

vniehues, 2022-01-21:
Hi there, 

I plan to add configurable BaseHeight & MaxHeight to my [HomeBridge Plugin](https://github.com/vniehues/homebridge-linak).
It would be nice if I could pass these values to your idasen-controller. 
The perfect way for me would be with argument flags like these:

`Idasen-controller --move-to 850 --base-height 600 --max-height 1300`

As it stands right now, when I try to send the desk down to lets say 580mm, your script would prevent that (because it's lower than the desk height).

What is your stance on this?

vniehues, 2022-01-21:
I think this would make your controller compatible with different Linak desks which could have different heights

rhyst, 2022-01-21:
Hey, I just merged a PR that adds a configurable base height. I will add an option to configure the movement range as well.

vniehues, 2022-01-21:
Oh no.. we were at it at the same time so I commented on that merged PR. Sorry for that.
Thank your for looking into the movement range! I'm looking forward to integrating it into the plugin :)

rhyst, 2022-01-21:
subraizada3 added this moments ago. I will make a release soon.

PR #35 (2022-01-21): Add movement range config option
Fix #34 

This may also be useful for compatibility with other Linak offerings. Many of them have a shorter range than the one used in the Idasen.
closed 2022-01-21 by rhyst