rhyst/linak-controller #36: Running as server and receiving values in client

vniehues, 2022-01-21:
Would it (in theory) be possible to have the server return the values to the client? 
That would allow us to just keep a server up for each desk and still let the clients handle value changes etc.

rhyst, 2022-02-07:
Hey, sorry for the delayed reply, can you elaborate a bit? Do you mean the the server should return the current desk height to the client as it is moving? Or just a single value at the end?

The latter may be possible with not too much effort as it looks like we don't close the request until the moving is completed but I'd have to look into how the asyncio server works.

I think updating as it moves would be trickier for the server as you'd need a web socket or something to push values back to the client, or implement some polling on the client side (and I don't know how the server handles multiple requests at the same time). 

I'm literally away from my desk at the moment 😁 so it will be a while before I can test anything.

vniehues, 2022-03-08:
Ahhh I'm sorry, I missed your previous answer! 
It was based on a small test I made where my plugin would keep a steady connection to allow for quicker responses. 
But it turns out that error handling was quite limited. 

I might give it another go now that the clients get a response from the server process.

Thank you for implementing it!

closed 2022-03-08 by vniehues