rhyst/linak-controller #41: Error: module 'asyncio' has no attribute 'exceptions'.

Nepomucene, 2022-02-19:
Fiddling with the controler.
If I use the app to wake up the controller, I can now move the desk up and down, to any height asked. It handles limits (soft and hard) graciously, but now throws the following at the end:
_ERROR:root:A message handler raised an exception: module 'asyncio' has no attribute 'exceptions'.
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/idasen_controller/main.py", line 246, in _move_to
    move_done.set_result(True)
asyncio.base_futures.InvalidStateError: invalid state_

rhyst, 2022-02-19:
I think this came up in a different issue but the person stopped responding: https://github.com/rhyst/idasen-controller/issues/28

What version of python are you using?

Nepomucene, 2022-02-19:
pi@Pi-Hole:~/.local/bin $ python --version
Python 2.7.16

rhyst, 2022-02-19:
Does replacing `asyncio.exceptions.InvalidStateError` with `asyncio.InvalidStateError` prevent that error?

Nepomucene, 2022-02-19:
Sorry, you're losing me here. No idea of how to do that...

Nepomucene, 2022-02-19:
Used the latest main.py and can confirm that on my raspi4, both errors seem to have disappeared (ascencio and bleak).
Desk still needs to be woken up by the app first to respond.

rhyst, 2022-02-19:
Ah okay. I will close this and we can track the other issue here: https://github.com/rhyst/idasen-controller/issues/32

Would you be able to try the code I suggested in that issue? https://github.com/rhyst/idasen-controller/issues/32#issuecomment-1046068950

rhyst, 2022-02-19:
Ah I see you did thanks!

closed 2022-02-19 by rhyst