rhyst/linak-controller #43: alias/rename/document this also works with brand name Linak controllers

ex-nerd, 2022-03-05:
I recently got an iMovr Lander desk, which uses the official Linak controller that is rebranded for Idåsen desks.  I am able to successfully connect and control the desk (once I realized that I needed UUID instead of MAC ... that callout in the readme isn't visible enough).

Yes, I will probably just create myself a symlink 😉.  Mostly I just wanted to suggest mentioning that this works with other Linak controllers.

rhyst, 2022-03-07:
I would like to do this but as I only have an Idasen desk I can't test on other devices and I don't want to make promises I can't keep. If others wanted to commit to working on ensuring support for other devices that would be appreciated and maybe a rename would be possible then.

Maybe for now I will just add a list of devices it reportedly works with.

ex-nerd, 2022-03-09:
See https://github.com/rhyst/idasen-controller/issues/44#issuecomment-1062687180

rhyst, 2022-03-11:
Note added about potential compatibility with other devices in the README.

closed 2022-03-11 by rhyst