rhyst/linak-controller #44: support multiple heights / automatic heights

ex-nerd, 2022-03-05:
I use 2 standing heights (with and without balance board), but this would also be nice for situations where a desk is shared by two people.  I imagine it wouldn't take much to make the height names in the config file generic so multiples could be added (`sit1`, `sit2`, `user1-sit`, etc).

My Linak DPG controller (looks identical to idåsen) also supports 4 saved memory points, so it would be cool if there was a way for `idasen-controller` to access these for positioning to them. It looks like https://github.com/anetczuk/LinakDeskApp (via https://github.com/anetczuk/linak_bt_desk/) supports these features.

rhyst, 2022-03-07:
Hey, I was thinking about this and I don't think it's necessary. It made me think that maybe the `--sit` and `--stand` commands are a bit pointless 😄 

If you are running this via a script you could just alias something like:

```
alias desk1='idasen-controller --move-to 1000'
alias desk2='idasen-controller --move-to 1200'
etc.
```

Maybe that's harder to do on windows?

Is there something I haven't thought about here? Just trying to keep the script simple :P

ex-nerd, 2022-03-09:
Not Windows here.  I can do aliases in my Mac (even call them via LaunchBar or another task manager).  In my case, this was more about my desire to maintain the height settings in the desk itself.  This may be a difference with the linak vs the idåsen controller, since I can set 4 different memory points in the desk -- that way I don't have to figure out the exact mm height of the desk and can just say "go to favorite #1" (for whatever value I've set that to this week). I mostly keep "sit" and "stand" favorites, but also "balance board" which is currently about 5cm taller than "stand" but changes depending on what I'm standing on, which keyboard wrist wrest I'm using (my desk has 2 computers on it), etc.  Still getting used to my new desk so maybe I won't be shifting these things around as much in the future (but for now, I'm still moving them around every day or two).

I'm currently trying to wrap my head around the command structure differences in `bleak` vs `bluepy` to see how feasible it would be to come up with something that merges your `bleak` stuff with the linux-only stuff that uses `bluepy` (which has a far more comprehensive set of documented commands). I'm long overdue learning `asyncio` so it's a good chance for a deep dive.  I'm starting with "desk capabilities" so I'm curious to see the differences between the two controllers.  Will keep you posted.

rhyst, 2022-03-11:
I have reconfigured this in `2.0.0` so now you can supply a list of favourites in your config file:
```
favourites:
    stand: 1000
    sit: 600
    middle 800
```
And move to them using the `move-to` command:

```
idasen-controller --move-to stand
idasen-controller --move-to sit
idasen-controller --move-to middle
```
And moving to heights still works:
```
idasen-controller --move-to 850
```

closed 2022-03-11 by rhyst