rhyst/linak-controller #46: Weird error message: No notify session started

erno1978, 2022-04-02:
Hi, I love the work you did. I am planning to use it for some nice home automation.

Right now I'm experiencing one problem. The desk is working, but most of the times (after moving the desk which works) it ends with:

Something unexpected went wrong: 
[org.bluez.Error.Failed] no notify session started

Do you know what I could do to prevent this error? Do you need other information? Thank you!

rhyst, 2022-04-10:
I don't know why that happens 😅 

As a start it would help to know:
- Operating system
- Python version
- Idasen controller version
- Is the device you're running on a raspberry pi?

But I won't be able to debug it unless you can modify and run the script from source (i.e. not using the pip installed version).

prawilny, 2022-05-01:
@rhyst 
It also happens on my setup:
- Fedora 36, Linux 5.17.3-302.fc36.x86_64
- Python 3.10.4
- idasen-controller 2.0.0 (downloaded by `pip3 install idasen-controller`)
- no, a normal desktop with the following bluetooth device `001 Device 004: ID 8087:0029 Intel Corp. AX200 Bluetooth`.

I checked out the current master, made the following change:
```
diff --git a/idasen_controller/main.py b/idasen_controller/main.py
index 1ed13b1..8d76edd 100755
--- a/idasen_controller/main.py
+++ b/idasen_controller/main.py
@@ -12,6 +12,7 @@ from bleak import BleakClient, BleakError, BleakScanner
 import json
 from functools import partial
 from appdirs import user_config_dir
+import traceback
 
 IS_WINDOWS = sys.platform == "win32"
 
@@ -455,7 +456,6 @@ async def forward_command(config):
     await ws.close()
     await session.close()
 
-
 async def main():
     """Set up the async event loop and signal handlers"""
     try:
@@ -477,6 +477,7 @@ async def main():
     except Exception as e:
         print("\nSomething unexpected went wrong:")
         print(e)
+        print(traceback.format_exc())
     finally:
         if client:
             print("\rDisconnecting\r", end="")
```
and run the program with `./main.py --config ~/.config/idasen-controller/config.yaml --move-to 820`.
I got (apart from the normal stdout) the following stack trace:
```
Traceback (most recent call last):
  File "/home/prawilny/idasen-controller/idasen_controller/./main.py", line 476, in main
    await run_command(client, config, print)
  File "/home/prawilny/idasen-controller/idasen_controller/./main.py", line 360, in run_command
    await move_to(client, target, log=log)
  File "/home/prawilny/idasen-controller/idasen_controller/./main.py", line 295, in move_to
    await unsubscribe(client, UUID_HEIGHT)
  File "/home/prawilny/idasen-controller/idasen_controller/./main.py", line 264, in unsubscribe
    await client.stop_notify(uuid)
  File "/home/prawilny/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 963, in stop_notify
    assert_reply(reply)
  File "/home/prawilny/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] No notify session started
```

erno1978, 2022-05-02:
> I don't know why that happens 😅
>
> As a start it would help to know:
>
> * Operating system
> * Python version
> * Idasen controller version
> * Is the device you're running on a raspberry pi?
>
> But I won't be [...]

Sorry, forgot to answer your questions.
My problems ceased to exist. Right now I'm using this setup:

* OS = Home Assistant OS 7.6
* Idasen version = github://j5lien/esphome-idasen-desk-controller@v3.0.0
* Yes, Raspberry Pi it is

It was quite a struggle, but right now it works as I want to. It's also pretty solid, no hiccups. So, thanks again!

rhyst, 2022-07-27:
I just published 2.0.1 which may fix this.

prawilny, 2022-07-27:
Yes, after the upgrade I can no longer reproduce the problem.

closed 2022-09-02 by rhyst