rhyst/linak-controller #47: Support for get_desk Height in server mode

radu-boboc, 2022-04-10:
Hello,

I've made a 3 button device, that connects to the desk, via one raspberrypi 2w. 
I am using your idasen-controller, to send commands to the desk ( and thank you very much for that).
However would like one feature, and that is to be able to print the desk height when running in server mode.
For example idasen-controller --forward --get_height I would like it to return the current desk height in mm.

Is it, please, possible to add it to the allowed commands?

I will mention that I am using the server mode, as a way to get a fast response time to the desk when I press a button and ignore the ever so annoying: 
`idasen-controller
Connecting failed
Device with address D6:48:E7:10:EF:27 was not found.`

rhyst, 2022-04-10:
I'm pretty sure this will happen if you just run `idasen-controller --forward`

radu-boboc, 2022-04-10:
you are correct.
I tried it before, but something didn't work. I am guessing it wasn't connecting to the desk.
So then problem solved :)
Thank you!

closed 2022-04-10 by radu-boboc