rhyst/linak-controller #50: Something unexpected went wrong or Adress not found

psentenac, 2022-07-26:
Hi

Issue : 
[image]
[image]

Pre-requisites :
[image]

[image]

[image]

[image]

[image]

Maybe wierd ?
1. If i'm not click on the desk bluetooth button (desk not appearing)
[image]
2. If i'm click on the button (desk appearing)
[image]

rhyst, 2022-07-26:
Thank you for the all the details 🙇

Just to confirm, is this an Idasen desk?

I am away for a couple of weeks do I may not be able to be very helpful.

Or1j1n, 2022-07-26:
> Just to confirm, is this an Idasen desk?
Yes

rhyst, 2022-09-02:
Is there no error message printed after "Something unexpected went wrong" ?

Or1j1n, 2022-09-15:
I dont have the issue anymore. Not sure why. But thanks anyway :P

closed 2022-09-15 by Or1j1n