rhyst/linak-controller #54: cant connect to the desk on multiple platforms

ctf0, 2022-09-02:
currently i have multiple os's on the same machine "windows & mac", if i tried to control the desk from one platform, the other wont work.

am not using the server option as its unrelaible, at least to the desk as the opend connection have to be forcefully killed other wise the desk will never accept connection to any other host including the phone app.

so is there anyway to get around that and make the desk connectable through mutiple platforms/devices ?

rhyst, 2022-09-02:
How are you running two OS? Dual booting or VMs or something else? 

What does unreliable mean?

There should be an error printed after `Something unexpected went wrong:`, and it would be helpful to see what that error is.

rhyst, 2022-09-02:
Where do you run the server? Do you run it on one OS and the reboot into the other?

I've not really tried using multiple devices so I wonder if this is more an issue with controlling from many devices (or OSs) rather than anything explicitly to do with dual booting.

Can you make sure you are running the latest version of `idasen-controller`? Should be 2.0.1.
```
>pip show idasen-controller
Name: idasen-controller
Version: 2.0.1
```

ctf0, 2022-09-02:
> Where did you run the server? Did you run it on one OS and then reboot into the other?

yes, exactly.

> Can you make sure you are running the latest version of idasen-controller? Should be 2.0.1.

yes it is, also having python `v 3.10.6`

rhyst, 2022-09-02:
I bumped the `bleak` version, maybe that will help. Try installing it with

```
pip install idasen-controller==2.0.2rc1
```

For what it's worth I tried using the official android app and the script:
- app wants the pair button pressed if the script was the last thing connected
- script fails if app is currently running
- script connects without pair button pressed even if the app was the last thing connected

So maybe make sure the android app isn't running (you didn't mention it so I assume it wasn't).

I also got an old laptop and tried it out and I was able to run commands on both one after the other with no repairing at all.  Both on windows though so maybe its the mac side of things that's not working?

ctf0, 2022-09-02:
> maybe that will help. Try installing it with

i will try and reply back, thanks.

> app wants the pair button pressed if the script was the last thing connected

yes, unless u disconnect the desk from the pc.

> script fails if app is currently running

yeah this is normal, because only one host can control the device.

> script connects without pair button pressed even if the app was the last thing connected

yeah same here, usually the issue is with the 2 platforms

> So maybe make sure the android app isn't running (you didn't mention it so I assume it wasn't).

yeah already done.

> I also got an old laptop and tried it out and I was able to run commands on both one after the other with no repairing at all. Both on windows though

maybe because both are windows & they are using different BT dongle/device, but i will try this with the 2 mac\s i have.

> so maybe its the mac side of things that's not working?

i thought that also but i have tried with another mac & the android phone to send and recieve files & it worked, also i can control the desk but not after windows has taken over.

rhyst, 2022-09-02:
Bleak is the library I'm using for the Bluetooth LE conneciton and I found something relevant in the Bleak documentation: https://bleak.readthedocs.io/en/latest/backends/macos.html

I wonder if the desk is only capable of being paired to one device at a time. On Windows and Linux the Bleak library automatically pairs whenever you try to connect and it is not paired but it does not on Mac, so you must manually re-pair. This also explains why the Android app requires re-pairing, but that using the script on two different Windows devices doesn't.

Can use the script straight away on Windows after using it on Mac (without re-pairing)?

ctf0, 2022-09-02:
okey, now am testing on windows & here is what am getting
> `idasen-controller --server`
```
Connecting failed
Could not get GATT services: Unreachable
```

> `idasen-controller --forward --move-to xxx`
```
Something unexpected went wrong:
Cannot connect to host 127.0.0.1:9123 ssl:default [The remote computer refused the network connection]
Unclosed client session
client_session: <aiohttp.client.ClientSession object at 0x00000286FD48B490>
```

also regarding the BT device it self.
- on mac, it doesnt persist after restart & for some reason it only pairs when i keep the `Bluetility` app opened, otherwise it wont connect, the device will just keep disconnectiong/reconnecting indefinitely.
- on windows the device is present in the BT devices list, but as u found out, it wont work if it was paired on mac, however i have the same behavior as the mac regarding the very quick disconnectiong/reconnecting using `Bluetooth LE Explorer`.

ctf0, 2022-09-02:
> I wonder if the desk is only capable of being paired to one device at a time

yeah this is normal for 99% of BT devices

rhyst, 2022-09-02:
Most of these errors are coming from the underlying Bleak library so you might have luck searching

The server mode just runs the same commands as running the script normally so I wouldn't expect running the server on the same device to do anything differently. I think the error with the forwarded command  happens because the server failed to start because it failed to connect.

> yeah this is normal for 99% of BT devices, for example u cant control the desk through phone & windows at the same time, they are not interchangable, u have to disconnect one of them so u can use the other.

I was referring to the desk being paired to multiple devices rather than being connected to the multiple devices which is a subtley different thing. Many bluetooth devices are capable of being paired to many devices at the same time.

> on windows the device is present in the BT devices list, but as u found out, it wont work if it was paired on mac

For me Windows always works without re-pairing, no matter what the previously connected device was. It is only the Linak Android app that requires re-pairing if something else was connected before. I do not have a Mac to test. You are saying you have to re-pair it on Windows as well?

What hardware are you running? Is it an Apple machine or other pc/laptop? Are you using a built in blueooth device or a dongle?

ctf0, 2022-09-02:
> I was referring to the desk being paired to multiple devices rather than being connected to the multiple devices

ah yes, u r correct.

> It is only the Linak Android app that requires re-pairing if something else was connected before

in my case, the android app is the lest glitchy of them 😆 

> What hardware are you running?

its a x99 desktop machine

> Is it an Apple machine or other pc/laptop ?

hackintosh, pc

> Are you using a built in blueooth device or a dongle?

dongle, TPLINK UB400

> PS:

i just downloaded the linak app for windows & and i can connect to the desk + having the same exact setup as the phone.
also changing the control between both is a matter of disconnecting a one & connecting the other.

rhyst, 2022-09-02:
I downloaded the Windows app and what I have said before is still true for me. For both the Windows and Android Linak app they require me to press the pair button if I have used a different program or device before them. The `idasen-controller` script never requires me to press the pair button and works as long as neither the Windows or Android Linak app is open.

And sorry, I am a bit confused by what works and what doesn't work for you so here are some yes/no questions to help me understand what is/isn't working for you:
- If you exclusively use Windows, does the `idasen-controller` command work consistently?
- If you use the Android Linak app to move the desk, then completely quit the app (I have to force stop it in app settings), then just run `idasen-controller` on Windows does it work without you pressing the pair button on the desk?
- If you use the Windows Linak app to move the desk, then completely quit the app (I have to close it from the taskbar), then just run `idasen-controller` on Windows does it work without you pressing the pair button on the desk?
- If you pair and run `idasen-controller` in Mac OS, and then boot into Windows, then just run `idasen-controller` does it work without you pressing the pair button on the desk?
- If you pair and run `idasen-controller` in Windows, and then boot into Mac OS, then just run `idasen-controller` does it work without you pressing the pair button on the desk?

ctf0, 2022-09-02:
> If you exclusively use Windows, does the idasen-controller command work consistently?

yes

> If you use the Android Linak app to move the desk, then completely quit the app (I have to force stop it in app settings), then just run idasen-controller on Windows does it work without you pressing the pair button on the desk?

yes

> If you use the Windows Linak app to move the desk, then completely quit the app (I have to close it from the taskbar), then just run idasen-controller on Windows does it work without you pressing the pair button on the desk?

yes

> If you pair and run idasen-controller in Mac OS, and then boot into Windows, then just run idasen-controller does it work without you pressing the pair button on the desk?

no

> If you pair and run idasen-controller in Windows, and then boot into Mac OS, then just run idasen-controller does it work without you pressing the pair button on the desk?

no

rhyst, 2022-09-02:
I think that might confirm my earlier idea. The desk can only be paired to one device at a time and the Bleak library handles re-pairing automatically on Windows, but not on Mac.

To confirm this you could try writing a wrapper script for Mac that pairs first and then runs `idasen-controller`. It seems like this is a program that claims to work on mac: https://github.com/toy/blueutil

Something like:

```
blueutil --pair <UID>
idasen-controller --mac-address <UID> --move-to 800
```

ctf0, 2022-09-02:
ok, will check it and get back to u

ctf0, 2022-09-03:
`blueutil` was useless, only https://github.com/jnross/Bluetility can do the discovering/pairing but sadly it doesnt have a cli, 

actually without `Bluetility` the mac bluetooth tool cant discover the desk, which is also true for the m1 mac i have.

closed 2022-09-03 by ctf0