rhyst/linak-controller #57: Locking computer stops table from moving

Slade01202, 2022-10-20:
First of all, thank you for this script. 

I use KDE running on Manjaro as well as the Linak DPG1C. The script runs fine until I lock my PC. After locking and unlocking, executing the script produces the following output:
```
Connected EA:69:XX:CE:XX:9C
Height:  641mm
Moving to favourite height: stand
Height:  641mm Speed:  0mm/s
Final height:  641mm (Target: 1116mm)
Disconnected 
```

I have only been able to fix it but connecting the phone app to my desk, moving it a centimeter, disconnecting, and executing the script again:
```
Connected EA:69:XX:CE:XX:9C
Height:  641mm
Moving to favourite height: stand
Height:  644mm Speed: 25mm/s
Height:  660mm Speed: 61mm/s
Height:  681mm Speed: 62mm/s
Height:  706mm Speed: 62mm/s
Height:  728mm Speed: 62mm/s
Height:  750mm Speed: 62mm/s
Height:  772mm Speed: 62mm/s
Height:  796mm Speed: 62mm/s
Height:  818mm Speed: 62mm/s
Height:  839mm Speed: 62mm/s
Height:  862mm Speed: 62mm/s
Height:  886mm Speed: 62mm/s
Height:  911mm Speed: 62mm/s
Height:  934mm Speed: 62mm/s
Height:  957mm Speed: 62mm/s
Height:  978mm Speed: 62mm/s
Height: 1001mm Speed: 62mm/s
Height: 1022mm Speed: 62mm/s
Height: 1045mm Speed: 62mm/s
Height: 1068mm Speed: 62mm/s
Height: 1091mm Speed: 62mm/s
Height: 1114mm Speed: 62mm/s
Height: 1116mm Speed:  0mm/s
Final height: 1116mm (Target: 1116mm)
Disconnected
```

rhyst, 2022-10-20:
If you lock and then unlock and then re-pair with the desk without using the phone, does that also fix it? Or not?

Slade01202, 2022-10-21:
Yes, that doesn't work either, unfortunately.

rhyst, 2022-10-21:
What about if you restart your computer? Is it just locking that causes this issue?

And do you happen to have a second device to test with? I.e. does it work if you run run idasen controller on some other device rather than using the phone?

mensfeld, 2023-04-24:
Similar issue here. Same solution: I can only get it back to work if I connect with my phone, move it a bit and then unpair the phone and connect to idasen back again

mensfeld, 2023-04-24:
```
Received command
Height:  762mm
Moving to height: 730
Height:  762mm Speed:  0mm/s
Final height:  762mm (Target:  730mm)
```

never moves

rhyst, 2023-04-24:
What about if you restart your computer?

And do you have a second device to test with? I.e. does it work if you run run idasen-controller on some other device rather than using the phone?

rhyst, 2023-04-24:
That is quite strange. Both reading characteristics (height/speed) and writing characteristics (commands like move) are both just BLE requests to the desk so if reading works then the controller is connected and capable of communicating with the desk. 

It suggests that somehow something happens to the desk when you lock your computer vs when you shut it down. I don't think there should be any communication between the desk and computer at that point so I don't know what is going on.

Some other questions if you are willing:
- What OS are you using?
- Are you using the "server" mode or just running the script standalone?
- Before locking your computer un-pair the desk, then lock, unlock and re-pair. Does the desk respond to move commands?
- If you have a laptop, what happens if you walk far out of BLE range (5-10 meters) and then come back after a few minutes (without locking)? Does the desk respond to move commands?
- If you have a laptop, what happens if you walk far out of BLE range (5-10 meters) and then lock your desk and then come back and unlock? Does the desk respond to move commands?

Trying to work out if your computer is maintaining some sort of connection to the desk while it is paired and locking messes it up somehow. Otherwise I don't know how the desk would "know" that your computer had been locked.

mensfeld, 2023-04-24:
1. Ubuntu 20.04
2. Server mode. After long inactivity and lock screen. I don't think that "locking" does that per se but rather something related to 3. IMHO power management on ubuntu USB. I disabled the USB suspend from Bluetooth and we will see once I have time to get back to this.
4. I don't use a laptop :( sorry

> Otherwise I don't know how the desk would "know" that your computer had been locked.

I don't think it's that. I think it's related to some sort of a faulty communication when Bluetooth is switched via USB suspend or something similar.

I also consider it weird because I can read position and I can observe it and speed when moving the desk manually.

I wonder if the wake up command is not sufficient / not fully waking up but :shrug:

mensfeld, 2023-04-24:
I disabled USB suspend and after a full restart I'm certain there's something "extra" that needs to be sent to the desk to activate it back.

mensfeld, 2023-04-25:
Didn't help. During the night it went down. That is, it started returning height 0 and then it started returning height 9123 :D my script pings it every minute. Will unlock and switch to a non-server mode and see what will happen then.

mensfeld, 2023-04-25:
I also don't think it's "locking" that causes this but it seems to be more related to persistent Bluetooth connection maybe or something like this.

Switched to non-server mode, we will see how it goes. I still track the metrics so it should operate ok.

mensfeld, 2023-04-25:
Same thing happens despite disconnecting working:

```
Moving the desk down in 15 seconds
Connected E6:8E:71:B5:6B:FD
Height: 1236mm
Moving to height: 750
Height: 1236mm Speed:  0mm/s
Final height: 1236mm (Target:  750mm)
```

no movement at all

rhyst, 2023-04-25:
How are you running `idasen-controller` ? Can you post the exact command/script?

mensfeld, 2023-04-25:
nothing fancy just moving up and down.

```
idasen-controller --move-to 750
```

now as it is stuck, if I pair it with my phone, unpair it, pair it with the computer, it can move like nothing happened. I still can get the position though with `--watch` :shrug: 

It's like the desk decides "ok, this Bluetooth device was moving me or sampling me too much, won't allow it to move the desk anymore'

rhyst, 2023-10-31:
@mensfeld does the DPGC1 update also fix this issue?

closed 2023-11-01 by rhyst

mensfeld, 2023-11-01:
@rhyst it does fix the issue :)