rhyst/linak-controller #60: PSA: Python 3.11 users need to use version 2.0.2rc3 of this package

prawilny, 2023-01-28:
Hi,
aiohttp 3.8.1, which is a dependency of 2.0.1 version marked as current on PyPI, has problems building on Python 3.11 - https://github.com/aio-libs/aiohttp/issues/6600. It's fixed in 3.8.3 of the library and in 2.0.2 version of this project.
Unfortunately, version 2.0.2 is marked as a `pre-release` on PyPI and thus `pip install idasen-controller` installs 2.0.1, which fails to install (the dependency cannot be built), so I had to spend a while checking the problem.
Maybe it's time to bump the current version to 2.0.2?

If you need a small test report: seems to be working on Fedora 37 with Python 3.11 and Ikea Idasen desk.

rhyst, 2023-02-06:
Ah thank you for the notice! I will update it as soon as I am back at my desk to test it (hopefully in a couple of weeks). Until then I will leave this issue open.

shirshir, 2023-03-07:
I ran into the same issue when installing it on Macos. You can install the pre-release using:

```
pip3 install idasen-controller --pre
```

rhyst, 2023-03-13:
2.0.2 is now published which updates bleak.

closed 2023-03-13 by rhyst