rhyst/linak-controller #68: Read/set favorite values from the desk provided from the Linak Desk Control app

TermoZour, 2023-09-02:
I just got an Ikea Idasen and I also installed the Desk Control app from Linak that is also compatible. The app allows you to set 3 predefined profiles and after that, you can use the controls on the desk to "snap" to those profiles while also allowing you to move between them. 

Would "setting" profiles be something you're interested in implementing? Maybe even getting those profiles from the desk itself (like you're reading the height) so they are always in sync.

Let's say the desk is at height 741. You have a set a profile from the app to height 758 and another at 1152. 
You pull the button up so you start raising the desk. It will keep rising until it reaches 758, then wait a second. If the button is still pressed, it will start rising again until it reaches 1152, at which point it will wait for a second again.

It would be really cool to have 3 profiles from the config sync with the 3 profiles set from the app, as well as being able to set new values from the CLI which can then be used with the hardware controls as well. This way, I can either control the desk and the profiles via it's hardware buttons, my personal laptop, or my phone when I'm using the work laptop.

TermoZour, 2023-09-11:
Found the code for an app that already seems to implement setting favorite values [here](https://github.com/anetczuk/linak_bt_desk/blob/7ac5fe1b69638976326842b27d76e52f0cc958fd/linak_dpg_bt/linak_device.py#L636)

I will test the app and follow along the lines to see if I can figure it out. I did send a packet with nRF Connect Android app but it seemed to clear all my favorites instead, so I might have done something wrong with the data I sent.

rhyst, 2023-10-31:
Thanks to the DPG1C issue I think I understand enough to implement this. Do you have suggestions for how it would work?

One idea, we have some special favourite names "favourite-1", "favourite-2" etc. which correspond to the favourite slots in the desk/app. 

If you do not set these in the idasen controller config then using them in the `--move-to favourite-1` argument will check the value from the desk first, and then move to it.

If you do set these values in the idasen controller config then using them in the `--move-to favourite-1` will move to that height _and_ write the value to the desk favourite.

TermoZour, 2023-10-31:
Honestly, this sounds like the best behavior.