rhyst/linak-controller #75: Cannot move desk to base height

mrzo, 2023-11-03:
Hi,
just noticed I cannot move my desk to the base height of 640mm (read from desk), "script ends with struct.error: ushort format requires 0 <= number <= 65535". Not sure if this is expected behavior, but from the description of the base_height option I think it isn't. Script version is latest 1.1.0 but same behavior on 1.0.3. Running it on Raspberry Pi 3, Raspberry OS (Bookworm), and on Linak's DPG1C controller. 

I suppose after implementing the kaml's fix this script works for me reliably for a few days. Previously I could only read from the desk, now I can move it nicely 👍🏼

pi@malina:~ $ linak-controller --mac-address ED:72:E6:E6:EB:0F --move-to 640
Connected ED:72:E6:E6:EB:0F
Capabilities: {'memSize': 4, 'autoUp': True, 'autoDown': True, 'bleAllow': True, 'hasDisplay': True, 'hasLight': True}
User ID: 01 2f d6 d1 2a f0 81 c9 61 14 68 c1 f6 19 2f 0b e8
Base height from desk:  640mm
Height:  900mm
Moving to height: 640

Something unexpected went wrong:
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.11/site-packages/linak_controller/main.py", line 210, in main
    await run_command(client)
  File "/home/pi/.local/lib/python3.11/site-packages/linak_controller/main.py", line 97, in run_command
    await Desk.move_to(client, target)
  File "/home/pi/.local/lib/python3.11/site-packages/linak_controller/desk.py", line 62, in move_to
    data = ReferenceInputService.encode_height(target.value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/linak_controller/gatt.py", line 81, in encode_height
    return bytearray(struct.pack("<H", int(height)))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: ushort format requires 0 <= number <= 65535

mrzo, 2023-11-03:
Yes, it does.

Cypher1, 2024-01-06:
My desk appears to move in 2mm increments when using `linak-controller` (see the output below)
But I can move it using the physical controls much more precisely (though it doesn't matter too much to me).

```
❯ linak-controller --forward --move-to 801
Height:  802mm Speed:  0mm/s
Height:  802mm
Moving to height: 801
Height:  802mm Speed:  0mm/s
Height:  802mm Speed:  0mm/s
Height:  802mm Speed:  0mm/s
Final height:  802mm (Target:  801mm)
~/.local/share/applications                                                                                                                                                                                                                                                     11:07:18 PM
❯ linak-controller --forward --move-to 800
Height:  802mm Speed:  0mm/s
Height:  802mm
Moving to height: 800
Height:  802mm Speed:  0mm/s
Height:  802mm Speed: -6mm/s
Height:  801mm Speed: -9mm/s
Height:  801mm Speed: -12mm/s
Height:  801mm Speed: -15mm/s
Height:  800mm Speed: -18mm/s
Height:  800mm Speed:  0mm/s
Height:  800mm Speed:  0mm/s
Height:  800mm Speed:  0mm/s
Final height:  800mm (Target:  800mm)
```
Watching while using the physical controls
```
❯ linak-controller --forward --watch       
Height:  800mm Speed:  0mm/s
Height:  800mm
Watching for changes to desk height and speed
Height:  800mm Speed:  0mm/s
Height:  800mm Speed:  6mm/s
Height:  800mm Speed:  9mm/s
Height:  800mm Speed:  0mm/s
Height:  801mm Speed:  6mm/s
Height:  801mm Speed:  0mm/s
Height:  801mm Speed:  5mm/s
Height:  801mm Speed:  8mm/s
Height:  801mm Speed:  0mm/s
Height:  801mm Speed:  7mm/s
Height:  801mm Speed:  0mm/s
Height:  801mm Speed:  6mm/s
Height:  801mm Speed:  1mm/s
Height:  801mm Speed:  0mm/s
Height:  802mm Speed:  6mm/s
Height:  802mm Speed:  8mm/s
Height:  802mm Speed:  0mm/s
```

I'll have a look to see if there's any obvious causes but I can't imagine it mattering too much :smile:

rhyst, 2026-09-22:
Maybe actually fixed in 1.3.3

closed 2026-09-22 by rhyst