rhyst/linak-controller #77: Experiencing `ValueError`: Characteristic notifications already started

dhruvkb, 2023-11-05:
`linak-controller` is raising the following error.

[image]

## Specs

- **Package name:** `linak-controller`
- **Package version:** 1.1.0
- **Python version:** 3.11.6
- **Installed via:** pipx
- **OS name:** macOS
- **OS version:** 14.1 (23B74)
- **Desk name:** Ikea Idåsen

snok3r-krk, 2023-11-30:
I have similar issue
[image]

Specs
Package name: linak-controller
Package version: 1.1.0
Python version: 3.11
Installed via: pip3
OS name: macOS
OS version: 14.1.1
Desk name: Ikea Idåsen

snok3r-krk, 2023-12-01:
It works with `1.1.1-dev` 🎈

rhyst, 2023-12-01:
Released 1.1.1

closed 2023-12-01 by rhyst