rhyst/linak-controller #85: query the state when in server mode

binlan, 2024-01-05:
Hi,

i think there is no way currently to get the current hight of the table when it runs in server mode, right? Ok, i could look into the logfile, but It would be nice to be able to curl (or linak-controller) the server for the current status. i run the linak-controller on a raspberry-pi in server mode mounted on the table. i now would like to integrate it in my home automation and  therefore it would be helpful to display the state.

what do you think.

thank you.

rhyst, 2024-01-05:
You can run:

```
linak-controller --forward
```

to get the current height

binlan, 2024-01-05:
ah, without everything. yes thats help.

thank you

rhyst, 2024-01-05:
No problem, I will leave this issue open as a reminder to put that in the README

binlan, 2024-01-05:
hmm, unfortunately it didnt do the job. if i e.g. move the table by hand, than a `linak-controller --server-address my-table --forward` move them back to the former position. I think it looks like, the server did not recognize the position change. If i would let my home automation poll with that method, it would make the hand switch unusable.

voruti, 2024-01-06:
I have connected my desk to my home automation over a REST Api and a (pushing) webhook, which I implemented in my own branch https://github.com/voruti/linak-controller/tree/typescript-rewrite. Maybe this can help you.

closed 2025-02-09 by rhyst

rpgdev, 2025-05-13:
> `linak-controller --forward`

Can we get this data _without_ running the server? The `watch` command returns it, but it lingers—it’d be awesome to have a command that just gets it and exits. It's what's missing to make this tool better for scripting. Crontab schedules are fine but dumb; I have a script that's smarter in handling this.

rhyst, 2025-05-14:
Does running `linak-controller` by itself not do what you are asking for?

rpgdev, 2025-05-14:
Yeah it does, found out after I posted. Output isn't too script friendly but I'd have to parse it either way. Thanks!