rhyst/linak-controller #87: Install of multidict failed → Update docs?

thomasaull, 2024-01-10:
I just tried to install `linak-controller` by running:
```bash
python -m pip install linak-controller
```
and got an error at the install of `multidict`. I don't have the error logs anymore, but it was the same or similar to this: https://forum.developers.symphony.com/question/1507074718375612416/building-wheel-for-multidict-pyproject-toml-error 

Anyway I solved it by installing the `python-devel` package. Maybe this could be added to the troubleshooting section of the docs?

rhyst, 2025-03-10:
I think I will leave this as an issue people can search as it seems to be quite a specific bug. Thank you for noting it though.

closed 2025-03-10 by rhyst

thomasaull, 2025-03-10:
@rhyst Sounds good to me! Thanks  for your work on `linkak-controller`, I use it almost every day :)