rhyst/linak-controller #90: Sending Commands to CBD6S

3e9b, 2024-04-02:
I have a Steelcase Ology which uses a Linak CBD6SP00020A-109. It has a Steelcase bluetooth adapter and I am able to connect successfully.

However using `--move-to ###` does nothing.

I would be happy to help diagnose and add support for this desk. Do you have any suggestions on how to proceed? The issue seems similar to #32. Looks like an error with sending commands.

```
Connected E5A13E83-F9E0-FC7F-6886-8ECCB0C8B30F
Capabilities: {'memSize': 4, 'autoUp': True, 'autoDown': True, 'bleAllow': False, 'hasDisplay': True, 'hasLight': True}
User ID: 01 e5 a1 3e 83 f9 e0 fc 7f 68 86 8e cc b0 c8 b3 0f
Base height from desk:  686mm
Height:  686mm
Moving to height: 700
Final height:  686mm (Target:  700mm)
Disconnected  
```

rhyst, 2024-04-13:
Hello! Sorry for the delayed response.

Can you describe the setup more? Its a Linak CBD6SP00020A-109 connected to some non-Linak bluetooth adapter? (And for clarity, the adapter is plugged into your desk, not your computer?). How do you normally control the desk?

A bit suspicious of the `'bleAllow': False` being reported in the capabilities. If that means that connecting via BLE is not allowed then its unlikely we can make it work with this script. On the other hand those capabilities are fetched via BLE so I'm not really sure if thats meaningful 😄

3e9b, 2024-04-22:
No worries. The bluetooth adapter is a Linak controller. Yes the adapter is plugged into the desk and not my computer. I normally control it with this adapter. It's sold by Steelcase as an [Active Touch Controller](https://steelcasehadsupport.zendesk.com/hc/en-us/articles/4417903423764-Active-Touch-Controller)

I can see the device info in Bluetility

The physical tag on the device says this:
```
Linak
Type: DPG1C72-000017
Item: DPG1C001-02
SW: 007088 ver 1.08
Model: DBG1C
```