rhyst/linak-controller #97: Package for python version 3.13

thomasaull, 2024-11-13:
Trying to install `linak-controller` with the following command:
```bash
python -m pip install linak-controller
```
gives me this error:
```
ERROR: Ignored the following versions that require a different python version: 1.0.0 Requires-Python >=3.8,<3.13; 1.0.1 Requires-Python >=3.8,<3.13; 1.0.2 Requires-Python >=3.8,<3.13; 1.0.3 Requires-Python >=3.8,<3.13; 1.1.0 Requires-Python >=3.8,<3.13; 1.1.1 Requires-Python >=3.8,<3.13; 1.1.1.dev0 Requires-Python >=3.8,<3.13; 1.2.0.dev0 Requires-Python >=3.8,<3.13
ERROR: Could not find a version that satisfies the requirement linak-controller (from versions: none)
ERROR: No matching distribution found for linak-controller
```
My OS hast python `3.13.0` installed:
```bash
python --version
Python 3.13.0
```
and it seems like this version is not supported yet.

rhyst, 2024-11-14:
Thanks for reporting. I will update this but I will not be able to for a few weeks. I would accept a PR in the meantime though.

thomasaull, 2024-12-02:
@rhyst I'm very inexperienced with python unfortunately and don’t have a development setup for it yet :-/

rhyst, 2024-12-03:
I have just released `1.2.0` which should add support for Python 3.13. Please let me know if this works for you 🙂

spoit, 2024-12-03:
Works again! Thx! (system: `Bluefin 41 (from Fedora Silverblue)`)

thomasaull, 2024-12-03:
Works perfectly on my system @rhyst thank you very much :)

closed 2024-12-03 by rhyst