rhyst/linak-controller #98: Installation with `uv` fails with error in `pyobjc-core`

dhruvkb, 2024-11-28:
When I try to install `linak-controller` using `uv tool`, I get a very long error message because `pyobjc-core` fails to build.

Somehow, `pipx` is able to install `linak-controller` so I suspect the difference may be Python 3.12 vs Python 3.13 and this issue might be merged with #97.

**Operating system:** macOS 15.1.1 (24B91) 
**Chip:** Apple M2 Pro

<details>
<summary>Log</summary>

```
~
❯ uv tool install linak-controller
Resolved 17 packages in 13ms
  × Failed to download and build `pyobjc-core==9.2`
  ╰─▶ Build backend failed to build wheel through `build_wheel` (exit status: 1)

      running bdist_wheel
      running build
      running build_py
      Overriding build_packages to copy PyObjCTest
      copying Lib/objc/_bridges.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_protocols.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience_sequence.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience_nsset.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_pycoder.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_lazyimport.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience_nsstring.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_framework.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience_nsdecimal.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/simd.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_informal_protocol.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_properties.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/__init__.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_callable_docstr.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_context.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_locking.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_descriptors.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_bridgesupport.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience_nsdictionary.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_pythonify.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience_nsarray.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience_nsobject.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_structtype.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_category.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_compat.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience_mapping.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_dyld.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_convenience_nsdata.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/objc/_transform.py -> build/lib.macosx-15.0-arm64-cpython-313/objc
      copying Lib/PyObjCTools/Signals.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTools
      copying Lib/PyObjCTools/TestSupport.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTools
      copying Lib/PyObjCTools/KeyValueCoding.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTools
      copying Lib/PyObjCTools/MachSignals.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTools
      copying PyObjCTest/test_voidpointer.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_specialtypecodes_unichar.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_fsref.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_methodedits.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_bundleFunctions.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_specialtypecodes_struct.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_dict_proxy.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_transform_integration.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_callbacks.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_unicode.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_weakref.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_transform.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_testsupport.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_protocol.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
      copying PyObjCTest/test_methods.py -> build/lib.macosx-15.0-arm64-cpython-313/PyObjCTest
[... 516 lines cut ...]
              This deprecation is overdue, please update your project and remove deprecated
              calls to avoid build errors in the future.
              ********************************************************************************

      !!
      Modules/objc/objc_util.m:850:23: error: call to undeclared function '_PyUnicode_EncodeUTF16'; ISO C99 and later do not support implicit
      function declarations [-Wimplicit-function-declaration]
        850 |         bytes_array = _PyUnicode_EncodeUTF16(pythonList, NULL, -1);
            |                       ^
      Modules/objc/objc_util.m:850:23: note: did you mean 'PyUnicode_DecodeUTF16'?
      /opt/homebrew/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/include/python3.13/unicodeobject.h:555:23: note:
      'PyUnicode_DecodeUTF16' declared here
        555 | PyAPI_FUNC(PyObject*) PyUnicode_DecodeUTF16(
            |                       ^
      Modules/objc/objc_util.m:850:21: error: incompatible integer to pointer conversion assigning to 'PyObject *' (aka 'struct _object *') from
      'int' [-Wint-conversion]
        850 |         bytes_array = _PyUnicode_EncodeUTF16(pythonList, NULL, -1);
            |                     ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      2 errors generated.
      error: command '/usr/bin/clang' failed with exit code 1

  help: `pyobjc-core` (v9.2) was included because `linak-controller` (v1.1.1) depends on `bleak` (v0.21.1) which depends on `pyobjc-core`
```

</details>

dhruvkb, 2024-11-28:
I tried installing with a `--no-build` rule and got this error message.

<details>
<summary>Log</summary>

```
~
❯ uv tool install --no-build linak-controller
  × No solution found when resolving dependencies:
  ╰─▶ Because pyobjc-core{platform_system == 'Darwin'}==9.2 has no usable wheels and building from source is disabled and only the following
      versions of pyobjc-core{platform_system == 'Darwin'} are available:
          pyobjc-core{platform_system == 'Darwin'}<=9.2
          pyobjc-core{platform_system == 'Darwin'}>10.0
      we can conclude that pyobjc-core{platform_system == 'Darwin'}>=9.2,<10.0 is incompatible.
      And because bleak==0.21.1 depends on pyobjc-core{platform_system == 'Darwin'}>=9.2,<10.0, we can conclude that bleak==0.21.1 cannot be used.
      And because only the following versions of bleak are available:
          bleak<=0.21.1
          bleak>0.22.0
      and all of:
          linak-controller<=1.1.0
          linak-controller==1.1.1
      depend on bleak>=0.21.1,<0.22.0, we can conclude that all of:
          linak-controller<=1.1.0
          linak-controller==1.1.1
       cannot be used.
      And because only the following versions of linak-controller are available:
          linak-controller<=1.1.0
          linak-controller==1.1.1
      and you require linak-controller, we can conclude that your requirements are unsatisfiable.

      hint: Pre-releases are available for `linak-controller` in the requested range (e.g., 1.2.0.dev0), but pre-releases weren't enabled (try:
      `--prerelease=allow`)
```

</details>

Then I managed to get it installed with the following command.

```
uv tool install --no-build --python=python3.12 linak-controller
```

rhyst, 2025-02-09:
This does seem to be fixed. Feel free to reopen if it is not.

closed 2025-02-09 by rhyst