rhyst/linak-controller #100: Server tries to move a desk even when not asked to

lastsecondsave, 2025-02-11:
If you start the app in a server mode and move a table:

```
$ linak-controller.exe --forward --move-to 800

Height:  795mm
Moving to height: 800
Height:  798mm Speed: 25mm/s
Final height:  800mm (Target:  800mm)
```

you won't be able to query a height anymore. Simple forward requests will try to move a table to the last set height:

```
$ linak-controller.exe --forward

Height:  800mm
Moving to height: 800
Nothing to do - already at specified height
```

My guess the rogue height comes from the global `config` object. Probably #85 is related.

rhyst, 2025-03-10:
Just pushed 1.3.0-dev which aims to solve this by removing the global config

lastsecondsave, 2025-03-10:
Thanks!

Currently it produces this exception:

```
Traceback (most recent call last):
  File "...\Python312\Lib\site-packages\aiohttp\web_protocol.py", line 462, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "...\Python312\Lib\site-packages\aiohttp\web_app.py", line 537, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "...\Python312\Lib\site-packages\linak_controller\main.py", line 178, in run_forwarded_ws_command
    await run_command(desk, command)
  File "...\Python312\Lib\site-packages\linak_controller\main.py", line 80, in run_command
    logger.log("Height: {:4.0f}mm".format(initial_height.human))
  File "...\Python312\Lib\site-packages\linak_controller\main.py", line 169, in log
    logger.log(message, end=end)
  File "...\Python312\Lib\site-packages\linak_controller\main.py", line 169, in log
    logger.log(message, end=end)
  File "...\Python312\Lib\site-packages\linak_controller\main.py", line 169, in log
    logger.log(message, end=end)
  [Previous line repeated 982 more times]
RecursionError: maximum recursion depth exceeded
```

Server is started with `linak-controller.exe --server`.